opentrack/opentrack #1043: WarThunder and Easy Anti Cheat

Twisted-WT, 2020-02-04:
War Thunder  http://www.warthunder.com  uses "Easy Anti Cheat" as a prevention solution. Opentrack works fine with WT when EAC is disabled, however when EAC is enabled, Opentrack no longer works in the game.

[2020_01_26_22_55_20__1640 (1).log](https://github.com/opentrack/opentrack/files/4152560/2020_01_26_22_55_20__1640.1.log)

The attached file should actually have .clog extension,  I renamed it so I could upload. The is a game file with EAC enabled and give a dll load error.

Many users are reporting this problem, however it is working for some players.

Currently I am using Windows 10. I have raised the issue with game developers as well at the EAC site. Help as you can imagine is  . . . .

Any guidance or assistance would be much appreciated.

sthalik, 2020-02-04:
> War Thunder http://www.warthunder.com uses "Easy Anti Cheat" as a prevention solution. Opentrack works fine with WT when EAC is disabled, however when EAC is enabled, Opentrack no longer works in the game.

They need to fix it on their end. I have no control over it.

Twisted-WT, 2020-02-04:
Thanks for the quick response.

Twisted-WT, 2020-02-06:
We have a response from EAC which I reprint here

==============

The "Untrusted System file"-error indicates that your Windows system is unable to verify the state of its files, either because the files in question are corrupted or Windows itself is. Any file that fails this basic check is prevented from loading into the game by the anti-cheat protection.

 

You can download sigcheck from https://docs.microsoft.com/en-us/sysinternals/downloads/sigcheck to check whether the file has a digital signature, as it is a common cause for files to be marked as untrusted. To do that first open the command prompt, then execute cd c:\users\user_name\downloads\sigcheck, assuming the unzipped sigcheck folder is located in your downloads. You will need to replace user_name with your system account name. Following that, without closing the command prompt, execute sigcheck.exe -i "C:\file_path\file_name.dll" (replacing file_path with the file's location and file_name with the name of the file you wish to check). If the result shows that the file is 'unsigned' this could indicate either of these two cases:

- the file is missing a digital signature, preventing it from being loaded into the game (applicable for files belonging to third-party software that is not part of Windows);
- your Windows CatRoot directory is corrupted somehow as it does not contain the signing information for that particular file.

While the first case can be resolved by obtaining a version of the file signed by an official root certificate authority (GlobalSign, VeriSign, etc.), the only sure way of fixing a corrupted CatRoot directory is to reinstall Windows. If you are already using a clean Windows installation we would recommend contacting Microsoft support for information on why this is happening and a possible solution.
========================

Please advise how to proceed

sthalik, 2020-02-07:
That doesn't seem to be the cause, no. The support person sent you a generic response without thinking about your problem.

Twisted-WT, 2020-02-07:
Thank you, I will update, once I have some more info on the issue

diVineProportion, 2020-02-12:
> War Thunder http://www.warthunder.com uses "Easy Anti Cheat" as a prevention solution. Opentrack works fine with WT when EAC is disabled, however when EAC is enabled, **Opentrack no longer works in the game**.

Not true. It no longer works for **_you_** and a few others. This is why I suggested that you gets get detailed hardware and software reports so you can compare your computer with another computer that doesn't work, then we can compare with my computer, one that does work and see if we can't find a common link.

Opentrack does work with EAC for many, just a few are having the issues. @Twisted-WT I think this is bit of information could mean a great deal for solving this issue, or it could mean nothing.

aadvark164, 2020-02-28:
I have had the same issue with no joy. The only way I could get head tracking to work is to use the version from the Trackhat website. This works with EAC on.

Twisted-WT, 2020-02-29:
@diVineProportion  it does not work for Anyone on Windows 7 . . .and so it seems not to work for those who have upgraded from windows 7.  Wipe out your shit and clean install again is not a solution.

@aadvark164   thanks . .sadly . .it needs a ps3 camera.

diVineProportion, 2020-02-29:
@Twisted-WT did I go over Dependency Walker with you or Jarcher_PL. Lookin through the clog files I noticed that it wasn't even attempting to be loaded, blocked by EAC I assume, but not record in their logs of this.

aadvark164, 2020-03-06:
> @diVineProportion it does not work for Anyone on Windows 7 . . .and so it seems not to work for those who have upgraded to windows 10 from windows 7. Wipe out your shit and clean install again is not a solution.
> 
> @aadvark164 thanks . .sadly . .it needs a ps3 camera.

Ok, so thats what I have, you can buy them on ebay and mod them yourself or buy them already modded. I did mine myself, took me 30 mins. What do you use on opentrack?
https://www.trackhat.org/trackhat-opentrack

sthalik, 2020-03-07:
I recommend FeelCustom -- <<https://vk.com/feelcustom>>. Best clip quality and clip supply with IR filter removal for both PS3 Eye variants.

Twisted-WT, 2020-03-07:
I made my own 3 point hat with white Leds, put a photo negative over an ordinary web cam lens . . .and BAM!! good to go.

diVineProportion, 2020-03-08:
@sthalik what happened to feelcustom? is there much stock left?

sthalik, 2020-03-08:
@diVineProportion no idea! Aren't they doing fine?

diVineProportion, 2020-03-09:
@sthalik when you said we it made me think you were a part of it. Mind you, this is google translate:

> FeelCustom ™ | Products for air and car simulators
> 29 Feb at 8:21 am
> Actions
> Hello, friends. We announce the termination of FeelCustom.
> 
> Despite this, FeelCustom will continue to fulfill all warranty obligations.
> 
> PS We still have some stocks in the warehouses, so for some time they will still sell deep fritters (13pcs), deep frit kits (10pcs), handbrake (4pcs). A certain reserve for warranty support (1 year) was also postponed.
> 
> Many thanks to everyone who has been with us all this time!

sthalik, 2020-03-09:
@diVineProportion I'm not working in the company. We designed the mini clip with @mrsanchos and subsequently shared the same design with TrackHat as well.

Twisted-WT, 2020-04-25:
So it seems, opentrack runs with Warthunder running in 64 bit mode and EAC, but not with Warthunder running is 32 bit mode.

zubaaan, 2021-09-07:
@Twisted-WT seems like the truth. No solutions yet?

Twisted-WT, 2021-09-07:
Apparently it seemed older processors have an older instruction set, that does not allow WT to run in 64bit mode. I bought a new rig,(Ryzen 5) and it started working just fine.

Twisted-WT, 2021-09-07:
Best close this . .

closed 2021-09-07 by Twisted-WT