opentrack/opentrack #1044: CMake build error

KulaGGin, 2020-02-07:
I am following the [quick start guide](https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows) and I get an error when I try to build:
```
Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.16299.
The C compiler identification is MSVC 19.24.28314.0
The CXX compiler identification is MSVC 19.24.28314.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
User settings file 'sdk-paths-Kulagin@MSVC-windows.cmake' doesn't exist
CMake Error at cmake/opentrack-qt.cmake:3 (find_package):
  By not providing "FindQt5Gui.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5Gui", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5Gui" with any
  of the following names:

    Qt5GuiConfig.cmake
    qt5gui-config.cmake

  Add the installation prefix of "Qt5Gui" to CMAKE_PREFIX_PATH or set
  "Qt5Gui_DIR" to a directory containing one of the above files.  If "Qt5Gui"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:76 (include)

Configuring incomplete, errors occurred!
See also "G:/Programs/CMake/bin/opentr/CMakeFiles/CMakeOutput.log".
```
And the log from `CMakeOutput.log`:
```
The system is: Windows - 10.0.16299 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler:  
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) Build Engine version 16.4.0+e901037fe for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 07.02.2020 12:09:07.
Project "G:\Programs\CMake\bin\opentr\CMakeFiles\3.16.4\CompilerIdC\CompilerIdC.vcxproj" on node 1 (default targets).
PrepareForBuild:
  Creating directory "Debug\".
  Creating directory "Debug\CompilerIdC.tlog\".
InitializeBuildStatus:
  Creating "Debug\CompilerIdC.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\CL.exe /c /nologo /W0 /WX- /diagnostics:column /Od /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"Debug\\" /Fd"Debug\vc142.pdb" /Gd /TC /FC /errorReport:queue CMakeCCompilerId.c
  CMakeCCompilerId.c
Link:
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\link.exe /ERRORREPORT:QUEUE /OUT:".\CompilerIdC.exe" /INCREMENTAL:NO /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:".\CompilerIdC.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:".\CompilerIdC.lib" /MACHINE:X64 Debug\CMakeCCompilerId.obj
  CompilerIdC.vcxproj -> G:\Programs\CMake\bin\opentr\CMakeFiles\3.16.4\CompilerIdC\.\CompilerIdC.exe
PostBuildEvent:
  for %%i in (cl.exe) do @echo CMAKE_C_COMPILER=%%~$PATH:i
  :VCEnd
  CMAKE_C_COMPILER=C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\cl.exe
FinalizeBuildStatus:
[... 112 lines cut ...]
Change Dir: G:/Programs/CMake/bin/opentr/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/MSBuild/Current/Bin/MSBuild.exe cmTC_b4263.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine version 16.4.0+e901037fe for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.24.28314 for x64

  CMakeCXXCompilerABI.cpp

  Copyright (C) Microsoft Corporation.  All rights reserved.

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_b4263.dir\Debug\\" /Fd"cmTC_b4263.dir\Debug\vc142.pdb" /Gd /TP /errorReport:queue "G:\Programs\CMake\share\cmake-3.16\Modules\CMakeCXXCompilerABI.cpp"

  cmTC_b4263.vcxproj -> G:\Programs\CMake\bin\opentr\CMakeFiles\CMakeTmp\Debug\cmTC_b4263.exe
```
Video demonstration for clarity showing that I have everything installed and set properly:
https://youtu.be/iyli4ycdhDw

I managed to build it by manually setting `Qt5Gui_DIR` cache variable to `G:\Programs\Qt\5.12.0\mingw73_64\lib\cmake\Qt5Gui` after I read about it [here](https://forum.qt.io/topic/96343/qt5guiconfig-cmake-etc-not-found-in-fresh-windows-installation/4#1) in the first answer.

Video demonstration of a fix:
https://youtu.be/MwjINV9gegY

I wanted to create an issue to ask for help but managed to fix it while I was writing this issue. Should I edit quick started guide, add there info about the fact that you have to include Qt5Gui_DIR, too, or did I miss something and there are other fixes for this?

sthalik, 2020-02-08:
>  Add the installation prefix of "Qt5Gui" to CMAKE_PREFIX_PATH or set
>  "Qt5Gui_DIR" to a directory containing one of the above files.  If "Qt5Gui"
>  provides a separate development package or SDK, be sure it has been
>  installed.

Exactly what it says on the tin. Set `Qt5Gui_DIR` to `qt-prefix/lib/cmake/Qt5Gui`.

Slion, 2020-09-01:
Could you fix this? I vaguely recall build was broken for Visual Studio 2019.

sthalik, 2020-09-15:
I'm unaware of the build being broken.