opentrack/opentrack #1047: Pitch Axis jumping

NoelShourai, 2020-02-11:
I recently installed the latest OpenTrack (2.3.11), and whenever I move the pitch axis halfway upwards the head tracking position makes a huge jump (suddenly adds 180 yaw, -90 pitch, 45 roll) and ends up in a position that looks very awkward; when I move it back down it returns to normal.

I thought at first that it was having difficulty tracking my LEDs when looking up but all three LEDs are picked up correctly and the jump is always consistent whenever I try to use more than 50% of the pitch axis upwards, no matter how I adjust the pitch sensitivity. I tried adjusting the sensitivity to block any output past 90 degrees but the jump always happens as long as I look upwards more than 50%, even if the output (in-game camera) is not 90 degrees or higher.

I already looked through the options and through the PointTracker settings but I didn't find anything that affects this behaviour. Is there a setting I'm overlooking?

I'm using a DelanClip, I've never had this problem in previous OpenTrack versions. Video shows the effect:
https://www.youtube.com/watch?v=nMvnebnTnac

sthalik, 2020-02-12:
Are you using dynamic pose in PT settings with 2.3.11-beta3?

NoelShourai, 2020-02-12:
Dynamic pose in PointTracker resolves the issue in beta3, but not in the latest version.

sthalik, 2020-02-14:
> Dynamic pose in PointTracker resolves the issue in beta3, but not in the latest version.

That's not how a clip behaves for almost everyone. You could improve the way it's mounted to solve this. The reason this doesn't work right now is that dynamic pose was forcefully disabled for clips. I'll reenable it for the next version.

commit 73ad388b (2020-02-14): tracker/pt: allow dynamic pose for clips

Fixes #1047
closed 2020-02-14 by sthalik