opentrack/opentrack #1048: opnetrack don't record camera setting

Verc0rs, 2020-02-12:
Hello,
In the camera setting, "Exposition" panel,  uncheck the box "auto" work but it's not recorded.
I'm using a "Logiteck QuickCam Pro 4000".
Trying with opentrack-2.3.11-beta3 to 2.3.11-20-g631c85c8.
Thanks

sthalik, 2020-02-12:
What camera setting exactly?

Verc0rs, 2020-02-13:
Maybe it's a misunderstanding on my part about how opentrack works.
In "input", I select "pointTracker", then "pointTracker setting". In the new windows, I click on "open" of the "camera setting (when available)" option. Is the new window that opens part of the "opentrack" software or does it use a Microsoft Windows 10 function?
In the second case the problem would be with the camera driver of windows 10.

sthalik, 2020-02-13:
> Is the new window that opens part of the "opentrack" software or does it use a Microsoft Windows 10 function?

Camera settings are provided by the driver. Try switching webcam driver to UVC Video.

Verc0rs, 2020-02-14:
Thanks for the answer, I was able to solve the problem.
Sorry for the inconvenience.

closed 2020-02-14 by Verc0rs