opentrack/opentrack #1050: Adding FreeTrack/TrackIR/SimConnect to input

ELXLabUF, 2020-02-14:
Hi, thanks for the amazing software,
I am wondering if there's any possibilities for OpenTrack to read in other program's output besides using UDP?
For example, I now see the FreeTrack/TrackIR/SimConnect options in output, any possibilites to add them into input as well?
What kind of work is needed for that?

sthalik, 2020-02-15:
What's the use-case for that?

zanven42, 2021-07-02:
an example use case.

I would like to migrate off windows to linux and having my existing TrackIR hardware work would be more ideal instead of buying something new.

memstat, 2022-05-30:
It would let you network stream any tracking systems not directly supported by opentrack, which can be useful if the original software becomes unmaintained and stops working (linuxtrack is already unmaintained). Someone could use a vm/pc with an old system to stream the protocol to another system, which is running the game. I don't know if that would be worth the trouble, though.

There may be other exotic scenarios where someone can't connect a tracking system not directly supported by opentrack to the computer running the game. Or simply to translate those systems to any of the outputs provided by opentrack, like mouse emulation.

human890209, 2022-11-05:
> What's the use-case for that?

I want to convert TrackIR into Joystick for a normal game.

sthalik, 2022-11-13:
Features for TrackIR hardware won't be supported. For more information, see https://naturalpointofview.blogspot.com/p/trackir-anticompetition.html.

closed 2022-11-13 by sthalik