opentrack/opentrack #1051: roadmap for 2.3.12

sthalik, 2020-02-24:
Roadmap:

- update LibOVR
- update RealSense SDK to R3

omegatao, 2020-02-26:
how about adding support for Nintendo Switch Joy-Con as an input device?

sthalik, 2020-03-29:
@omegatao not unless someone contributes the code.

Need4SpeedOG, 2020-04-28:
RealSense SDK is now open sourced at 2.0 also on GitHub https://github.com/IntelRealSense/librealsense don't know if this would be a major retooling

sthalik, 2020-04-28:
The open-source version is missing face-tracking. It can't be used.

sthalik, 2020-05-08:
> Is there anything that can be done to make SteamVR + Vive tracker without an HMD work?

Only if steamvr allows it.

PhilWilliams-dev, 2020-05-08:
> > Is there anything that can be done to make SteamVR + Vive tracker without an HMD work?
> 
> Only if steamvr allows it.

What's interesting is that there is a way to fire up steamVR without the HMD by using the 'null' driver which works, I get green tracking stations and the Vive tracker but for some reason Opentrack does not receive any tracking data unless the physical HMD is connected...

sthalik, 2020-05-31:
I'm almost ready to release a ps3eye open driver build. If you want to test it, please:

- Install `libusbK.sys` as the driver for the camera, by youself (without me providing instructions).
- Reply in this issue.

sthalik, 2020-07-01:
FWIW, I'm so happy with the state of the driver that I might just as well push a release candidate. The CPU usage on Windows is very minimal.

wostg, 2020-08-11:
Hey fella, it occurred to me while I'm throwing money around (1 week till FS2020 lol) I should support software that will get equivalent use. (kinect working great on the fs beta). I saw the note about supporting the EFF but if you or other significant contribs have one of those 'buy me a coffee' deals, or other currently preferred charities please do say.

wostg, 2020-08-11:
And you Slion! 👍

closed 2022-10-21 by sthalik