opentrack/opentrack #1052: Questions about NPClient.dll

slavnikname, 2020-02-26:
Hello Stanislaw!

Did you write the library "modules\NPClient.dll" and the file "modules\TrackIR.exe" by yourself?
If so, then what documents and rules did you follow?
Does the TrackIR.exe file just start to present in the operating system and does nothing?

My questions are related to this: I made a head position tracking device that connects to a PC via USB and works like a joystick.
But the game IL-2 Sturmovik does not allow the joystick to control the shifts of the view in the cockpit.
To use my device in this game, I have to cast the joystick to TrackIR using OpenTrack.

There is a plan to write my own little OpenTrack-like process for such a broadcast.
But there is another idea - write my own version of NPClient.dll that will directly read the joystick HID-collection object instead of reading the "FT_SharedMem" file mapping object.
In your opinion, is it possible?

sthalik, 2020-02-26:
> But there is another idea - write my own version of NPClient.dll that will directly read the joystick HID-collection object instead of reading the "FT_SharedMem" file mapping object.
> In your opinion, is it possible?

Yes, although time-intensive. Incidentally, opentrack already supports this.

slavnikname, 2020-02-26:
Thanks! 
And the first questions will be answered?

sthalik, 2020-02-26:
> Did you write the library "modules\NPClient.dll" and the file "modules\TrackIR.exe" by yourself?
If so, then what documents and rules did you follow?

It's not documented but it's not hard, either. See `contrib/trackir-client` for more details on the protocol.

slavnikname, 2020-03-11:
Thanks! 
But it might be better to look contrib\very-important-source-code\npclient.c or contrib\npclient\npclient.c ?

sthalik, 2020-03-11:
I cleaned up the contrib structure in 2.3.11.

slavnikname, 2020-03-22:
Hello Stanislaw!

More questions arose:

What is the reason for blocking the modifications of NPClient.dll whith battleye anti-cheat?
The fight for NaturalPoint's rights? Or the equalization of network players?
Can you hope that battleye developers will continue to be loyal to the old version of NPClient.dll from Opentrack?
What happens if battleye completely kills third-party modifications of NPClient.dll?

With the help of your sources, I have made my modification of NPClient.dll and my device can now fully work in IL2.
In ArmA3, it can also work through this method, but without launching battleye.
Apparently it will also be in many other games.
It’s good that ArmA3 fully implements control via the free joystick interface.
Unlike IL2, it is also probably a reverence towards NaturalPoint.

sthalik, 2020-03-29:
> What is the reason for blocking the modifications of NPClient.dll whith battleye anti-cheat?

A library can facilitate cheating.

> The fight for NaturalPoint's rights?

No. BattlEye specifically allows opentrack's library. That happened without me asking for it.

> In ArmA3, it can also work through this method, but without launching battleye.
> Apparently it will also be in many other games.
> It’s good that ArmA3 fully implements control via the free joystick interface.
> Unlike IL2, it is also probably a reverence towards NaturalPoint.

This is blatantly untrue.

> With the help of your sources, I have made my modification of NPClient.dll and my device can now fully work in IL2.

Interesting, what are you doing?