opentrack/opentrack #1053: Does it have Trackir support for linux ?

sigzegv, 2020-02-28:
I have a trackir 5 and I was wondering if it could work with opentrack (and eventually a wine connector), and if yes, how.
I tried the prebuilt package on archlinux but no inputs seems to match my need. I see some contrib related to trackir but with exe files only ? I tried to build it from sources but same result..

Why I would want to know if it's supported here, it's because actually with linuxtrack-wine I'm limited to wine 3.21, any version greater than 4 make their wine plugin crash, they don't answer support requests, and it seems not developped anymore...

sthalik, 2020-02-28:
opentrack exposes NPClient data pretty much the same as on Windows -- see the Wine protocol.

sigzegv, 2020-02-28:
I don't see any documentation on the proto wine section, neither in the wiki. Some configuration instructions could be appreciable.

sthalik, 2020-02-28:
[image]

sthalik, 2020-02-28:
[image]

sigzegv, 2020-02-28:
Sorry but I don't understand what you are showing to me without some explanation :disappointed: .

I don't have anything related to wine in the opentrack interface, no inputs nor outputs, and no options related (for now I have a default installation).

I see the NPClient dlls in the libexec dir (and a TrackIr.exe file). but nothing explains me how to use them.

sthalik, 2020-02-29:
Select `SDK_WINE` (see `ccmake` and `cmake-gui`).

sigzegv, 2020-02-29:
Ok that's fine I managed to build a wine output, but I don't know how to get the trackir input.
Not sure if it's point tracker but it doesn't work, I didn't see any options related to trackir in the cmake parameters.

sthalik, 2020-02-29:
Wine output works with TrackIR and Freetrack.

sigzegv, 2020-02-29:
I mean about inputs in opentrack. I should have a compatible input device in opentrack right ?

I don't see trackIr or freetrack in inputs, which I have are : Easy Tracker, Freepie, Fusion, Hatire Arduino, Point tracker, S2Bot, Test tracker, UDP over network.

sthalik, 2020-02-29:
You can't use a TrackIR camera as input.

sigzegv, 2020-03-01:
I don't understand, why didn't you told me that at first when as was explaining I was using a track Ir 5 under linux ?

sthalik, 2020-03-01:
opentrack supports *output* using the same protocol as TrackIR. Just not input.

closed 2020-03-01 by sigzegv