opentrack/opentrack #1056: Bind keys to user func

WF000, 2020-03-01:
Hi Halik, hope you're well.

It is not an issue, just a question and/or possible support .
Is it possible and how to bind some keys to "Custom center pose"  user-func , via .ini or some configuration script?
There are enough free binds and those could be  (and ARE) quite useful , as it allows customizing center position on-the-fly from client application .
Sorry for big picture.

Thanks
WF

[image]

sthalik, 2020-03-07:
Not until opentrack allows scripting.

sthalik, 2023-11-05:
My luajit fork could use some testing: https://github.com/sthalik/luajit