opentrack/opentrack #1057: Connecting to UDP server

AndrisJ, 2020-03-01:
I have setup DualShock 4 controller with DS4Windows which has UDP server providing pitch, roll and yaw data from the DS4 gyro sensor. 
I see that there are FreePIE and UDP over network but those are listeners for the incoming data, instead I would like to use opentrack to connect to an existing server.
Is there a way to connect opentrack to an existing UDP server for the data?

sthalik, 2020-03-01:
Yes, use the UDP output.

AndrisJ, 2020-03-01:
I meant connect to external server to receive data (Input) rather than sending data.
Receiving end is the freetrack 2.0 Enhanced for me (tested with phone and FreePIE received).

sthalik, 2020-03-01:
By virtue of the UDP protocol, there's no way to connect in order to receive data.

rvt, 2020-03-07:
Andris,

I would write a simple Lua script to do the work for you.
Lua has socket.udp that can do the work for you.
You can then receive the UDP datagrams from DS4Windows and send them to opentrack.

arifinmnur, 2021-01-29:
Hi, is problem solved?

Markd-One, 2021-02-07:
Not actual anymore, can be closed.