opentrack/opentrack #1058: Compiling opentrack on linux

SGD1953, 2020-03-01:
Hi,

I'm trying to compile opentrack on linux (ubuntu 16.04) and failing miserably. I understand this is probably not a bug as such, but it might be helpful for other users.

I tried to follow the tutorial at https://tkoletivrijedi.in.rs/node/219?language=en but ran into a number of issues. First of all. I did not have QT installed (that I could have figurted from the docs though). Once this was sorted, I found that the C/C++ compiler was not current enough so I updated to the latest & greatest. Via `ccmake` I updated all paths to point to QT5.14 & clang 9.0.0 binaries.

But now I'm stuck and lacking deeper C expertise I cannot get any further (for version `2.3` & `unstable`).

```
...
[ 27%] Building CXX object api/CMakeFiles/opentrack-api.dir/plugin-api.cpp.o
[ 28%] Built target i18n-module-tracker-udp
[ 29%] Built target opentrack-version
[ 29%] Building CXX object api/CMakeFiles/opentrack-api.dir/moc_plugin-api.cpp.o
[ 30%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/base-path.cpp.o
[ 30%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut.cpp.o
[ 30%] Built target i18n-lupdate
[ 30%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/camera-names.cpp.o
In file included from /home/xyz/dev/opentrack23/build/api/moc_plugin-api.cpp:10:
In file included from /home/xyz/dev/opentrack23/build/api/../../api/plugin-api.hpp:11:
In file included from /home/xyz/Qt/5.14.1/gcc_64/include/QtCore/QString:1:
/home/xyz/Qt/5.14.1/gcc_64/include/QtCore/qstring.h:1514:29: error: cannot initialize a parameter of type 'wchar_t *' with an rvalue of type 'const wchar_t *'
    str.resize(toWCharArray(str.data()));
                            ^~~~~~~~~~
```
Could it be that I'm still using some wrong compiler version (using `clang version 9.0.0 `)? Could it be too new?

Any help would be greatly appreaciated, happy to document my steps once I can get it compiled for the greater benefit.

Thank you

sthalik, 2020-03-02:
We're probably talking about this piece of code:

```c++
inline std::wstring QString::toStdWString() const
{
    std::wstring str;
    str.resize(length());
#if __cplusplus >= 201703L
    str.resize(toWCharArray(str.data()));
#else
    if (length())
        str.resize(toWCharArray(&str.front()));
#endif
    return str;
}
```

It works for me with clang 9.0.1 and Qt 5.14.1.

SGD1953, 2020-03-02:
Thank you so much for the fast reply!

I installed clang 9.0.1 and tried again, same result. Now I'm a bit lost, if we use the same source & the same compiler - why is my compiler complaining then?

sthalik, 2020-03-02:
Try replacing

```c++
#if __cplusplus >= 201703L
```

with

```c++
#if 0
```

SGD1953, 2020-03-02:
That works but it leads into tons of other errors, so the compiler wouild have to be very broken. I tripple-checked all paths via ccmake & they point to the right locations as far as I can tell.

```
In file included from /home/xyz/dev/opentrack23/compat/math.hpp:3:
/home/xyz/dev/opentrack23/compat/macros.hpp:23:41: error: template argument for non-type template parameter must be an expression
using cv_qualified = std::conditional_t<std::is_fundamental<remove_cvref_t<t>>,
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:2526:17: note: template parameter is declared here
  template<bool _Cond, typename _Iftrue, typename _Iffalse>
                ^
In file included from /home/xyz/dev/opentrack23/compat/correlation-calibrator.cpp:3:
/home/xyz/dev/opentrack23/compat/math.hpp:52:15: error: no template named 'cv_qualified'
    using w = cv_qualified<decltype(val + min + max)>;
              ^
/home/xyz/dev/opentrack23/compat/math.hpp:53:31: error: use of undeclared identifier 'w'
    return util_detail::clamp<w>::clamp_(val, min, max);
                              ^
/home/xyz/dev/opentrack23/compat/math.hpp:57:52: error: no template named 'is_floating_point_v' in namespace 'std'; did you mean 'is_floating_point'?
inline auto iround(t val) -> std::enable_if_t<std::is_floating_point_v<remove_cvref_t<t>>, int>
                                              ~~~~~^~~~~~~~~~~~~~~~~~~
                                                   is_floating_point
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:350:12: note: 'is_floating_point' declared here
    struct is_floating_point
           ^
In file included from /home/xyz/dev/opentrack23/compat/correlation-calibrator.cpp:3:
/home/xyz/dev/opentrack23/compat/math.hpp:57:47: error: template argument for non-type template parameter must be an expression
inline auto iround(t val) -> std::enable_if_t<std::is_floating_point_v<remove_cvref_t<t>>, int>
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:2522:17: note: template parameter is declared here
  template<bool _Cond, typename _Tp = void>
                ^
In file included from /home/xyz/dev/opentrack23/compat/correlation-calibrator.cpp:3:
/home/xyz/dev/opentrack23/compat/math.hpp:63:52: error: no template named 'is_floating_point_v' in namespace 'std'; did you mean 'is_floating_point'?
inline auto uround(t val) -> std::enable_if_t<std::is_floating_point_v<remove_cvref_t<t>>, unsigned>
                                              ~~~~~^~~~~~~~~~~~~~~~~~~
                                                   is_floating_point
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:350:12: note: 'is_floating_point' declared here
    struct is_floating_point
           ^
In file included from /home/xyz/dev/opentrack23/compat/correlation-calibrator.cpp:3:
/home/xyz/dev/opentrack23/compat/math.hpp:63:47: error: template argument for non-type template parameter must be an expression
inline auto uround(t val) -> std::enable_if_t<std::is_floating_point_v<remove_cvref_t<t>>, unsigned>
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:2522:17: note: template parameter is declared here
  template<bool _Cond, typename _Tp = void>
                ^
[ 34%] Built target opentrack-api
[ 34%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut.cpp.o
In file included from /home/xyz/dev/opentrack23/qxt-mini/qxtglobalshortcut_x11.cpp:48:
In file included from /home/xyz/dev/opentrack23/qxt-mini/powerset.hpp:3:
/home/xyz/dev/opentrack23/compat/macros.hpp:23:41: error: template argument for non-type template parameter must be an expression
using cv_qualified = std::conditional_t<std::is_fundamental<remove_cvref_t<t>>,
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:2526:17: note: template parameter is declared here
  template<bool _Cond, typename _Iftrue, typename _Iffalse>
                ^
/home/xyz/dev/opentrack23/compat/correlation-calibrator.cpp:52:22: error: cannot initialize a variable of type 'const double' with an rvalue of type 'void'
        const double val = clamp(data[k], min[k], max[k]);
                     ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/xyz/dev/opentrack23/qxt-mini/qxtglobalshortcut_x11.cpp:48:
/home/xyz/dev/opentrack23/qxt-mini/powerset.hpp:17:24: error: no template named 'is_integral_v' in namespace 'std'; did you mean 'is_integral'?
    static_assert(std::is_integral_v<size_type_>, "");
                  ~~~~~^~~~~~~~~~~~~
                       is_integral
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:322:12: note: 'is_integral' declared here
    struct is_integral
           ^
In file included from /home/xyz/dev/opentrack23/qxt-mini/qxtglobalshortcut_x11.cpp:48:
/home/xyz/dev/opentrack23/qxt-mini/powerset.hpp:17:49: error: expected '(' for function-style cast or type construction
    static_assert(std::is_integral_v<size_type_>, "");
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
[... cut ...]
                  ~~~~~^~~~~~~~~
                       is_same
/usr/lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/type_traits:1517:12: note: 'is_same' declared here
    struct is_same
           ^
/home/xyz/dev/opentrack23/qxt-mini/qxtglobalshortcut_x11.cpp:367:81: error: expected '(' for function-style cast or type construction
    static_assert(std::is_same_v<xcb_key_press_event_t, xcb_key_release_event_t>);
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
```

sthalik, 2020-03-02:
You must upgrade `gcc` since clang tries to use ancient libstdc++.

SGD1953, 2020-03-02:
Thank you so much, that was indeed the issue. After upgrading `gcc` to 9.2 it finally compiled!

closed 2020-03-02 by SGD1953