opentrack/opentrack #1059: 2.3.11 Window size difference

JarcherPL, 2020-03-04:
The newest version looks weird on my PC. Like this:

[image]
[image]

MathijsG, 2020-03-31:
That is by design. Although I understand the confusion.

sthalik, 2020-03-31:
That's not by design. Can you see whether the build from <<http://opentrack.misaki.pl>> works?

MathijsG, 2020-03-31:
Well I have the exact same thing then. With a fresh build.

MathijsG, 2020-03-31:
Still looking different than the earlier versions. It looks like this with the build you linked:
[image]

sthalik, 2022-09-30:
Can you confirm this with [opentrack 2022.3.0](https://github.com/opentrack/opentrack/releases/tag/opentrack-2022.3.0)?

closed 2024-06-07 by sthalik