opentrack/opentrack #1060: 2.3.11 Shift+C hotkey problem

JarcherPL, 2020-03-04:
For older versions I used Shift + C for centering.
Shift + Z = start/stop
Shift + X = pause

In 2.3.11 Shift + C stops "input data", but "raw tracker data works". To make it work again tracking needs to be restarted.
Rarely it crashes the program.

sthalik, 2020-03-07:
You're saying that center functionality causes the data to stay at zero?

JarcherPL, 2020-03-07:
Not exactly, only "Shift + C". When I change centering for some other hotkey it works normally.
It happens only in 2.3.11, so im using 2.3.10

marccreal, 2020-07-26:
I also experienced this problem. Had it binded to CTRL+\.
If I bind it to a single key, I did not experience the bug. So it seems to be connected to a two-key-binding.

sthalik, 2020-09-15:
Can you confirm if it works in 2.3.12?

marccreal, 2020-09-16:
Yes, I can confirm!

JarcherPL, 2021-09-21:
I confirm it is working good now. Error is not present in 2021.1.2
Sorry for late response.

closed 2021-09-21 by sthalik