opentrack/opentrack #1061: Inertial tracker zero crossing

vostrenkov, 2020-03-09:
Hi!
There is a problem when input from inertial device (hatire arduino) crossing from 179 degrees to +180 degrees. Octopus makes a full circle rotation to opposite direction in this case. 
This becomes a problem if inertial device is used for VR application where full 360 degree rotations are required

sthalik, 2020-03-09:
Have you tried disabling the filter?

vostrenkov, 2020-03-09:
Filter disabled and response curves set to linear 180/180. 
When input crossing from 180 to -180 output jumps from 180 to -40. If input is -140 output is 0.

sthalik, 2020-03-10:
Which tracker is it?

vostrenkov, 2020-03-10:
This one:
https://forum.il2sturmovik.ru/topic/7524-5dof-%D0%B8%D0%BD%D0%B5%D1%80%D1%86%D0%B8%D0%BE%D0%BD%D0%BD%D0%B8%D0%BA-%D1%81%D0%BE-%D1%81%D0%BC%D0%B5%D1%89%D0%B5%D0%BD%D0%B8%D1%8F%D0%BC%D0%B8/

The issue about zero crossing is complained by its developer GO63

sthalik, 2020-03-10:
OK. What's the raw data before and after the transition?

vostrenkov, 2020-03-10:
We have made videos to show it in better way:

2.3.10 without accela filter - output jumps at -180 to 180 input transition:
https://youtu.be/kkt5UNiem2c

2.3.11 with accela filter - better but jumps at output -180 to 180 transition
https://youtu.be/xaVfckm1Oe4

2.3.11 without filter - works good (sorry, we tested at 2.3.10 before and it didn't work without filter)
https://youtu.be/C_vBCQF-TCA

sthalik, 2020-03-11:
What's the issue with the last video? It appears to work as it should.

vostrenkov, 2020-03-11:
Last is ok but there is no filter. So you cant adjust deadband and smoothing. 
2.3.11 is quite good, but is there any chance that filter will handle transition properly?

sthalik, 2020-03-11:
A filter won't work regardless of circumstances. There's a singularity on 0/180.

vostrenkov, 2020-03-11:
Ok, roger that. Thank you anyway!

closed 2020-03-11 by vostrenkov