opentrack/opentrack #1063: Opentrack not starting

DanFraserUK, 2020-03-16:
I'm not sure what has happened here to be honest!  A few days ago it worked, now apparently an entire .so.7 has vanished and the package that contains that is installed (and reinstalled!).

`$ opentrack`
`opentrack: error while loading shared libraries: libprocps.so.7: cannot open shared object file: No such file or directory`

I'm using Manjaro

`$ uname -r`
`5.5.6-arch1-1-fsync`

I'm stuck as to what to try next, any ideas?

sthalik, 2020-03-17:
Extract the file from the Debian package.

DanFraserUK, 2020-03-17:
I reinstalled the AUR package, nothing, rebooted, it worked.

I'm still confused as to what caused this, but it's fixed so I guess it can be closed!

closed 2020-03-17 by DanFraserUK