opentrack/opentrack #1064: Opentrack point tracker can track me super smooth, but game lags behind

eclair4151, 2020-03-17:
I'm not sure where the issue is lying, and if it's my computer, p3d, or this, but figured i would ask. I set up point tracker and it is tracking me super well and fast with the octopus. But when i try to use it in a game it lags behind by about .5 seconds which makes it super nauseating and unplayable. I'm not sure if there is some setting or something i am missing?
https://photos.app.goo.gl/9MgkBfM1X5qWv3gR8

MathijsG, 2020-03-20:
What framerate do you get without the tracking? And what framerate wíth the tracking? And try your filter settings maybe?

And what Hertz setting do you use in Opentrack camera settings?

sthalik, 2020-03-29:
> But when i try to use it in a game it lags behind by about .25 seconds which makes it super nauseating and unplayable.

What game is this?

eclair4151, 2020-03-30:
Its prepar3d, and I will get some screenshots of framerate and my settings later today, but i have a radeon vii graphics cards and i9-9900k, so im like %100 percent sure performance isnt an issue.

sthalik, 2020-03-30:
Paste the contents of the filter settings window.

eclair4151, 2020-03-31:
Here are the settings windows, as well as another video, I also wanted to ask another question, for some reason when  i turn my head and the right or left LED cross paths with the top one, it gets very confused which is the top led and spins the camera. Any way to avoid this?
[image]
[image]

https://photos.app.goo.gl/46dienyMFWDZLb396

sthalik, 2020-03-31:
> when i turn my head and the right or left LED cross paths with the top one

Unfortunately this is normal. The tracker has no clear way to otherwise distinguish the LEDs from one another. If this is a problem, use the "dynamic pose" option, which comes with its own drawbacks.

sthalik, 2020-03-31:
See if the delay exists in other games, for instance Falcon BMS.

MathijsG, 2020-03-31:
If you want smoother movement you could also upgrade to a PlayStation 3 eye camera. It can do 75 Hertz at 640 * 480 or 120 at 240 * 320. Or maybe your webcam can also have higher frame rate? What if you set the 30 to 60 for example, what actual frame rate do you get?

eclair4151, 2020-03-31:
Looks like my webcam is only 30fps
https://www.logitech.com/en-us/product/hd-webcam-c310
"At 30 frames per second, video quality is smooth, while images are crisp, colorful and contrasted"

But my intuition tells me a faster camera frame rate would not fix the problem, as that doesn't explain why the octopus can track me super quickly, but the game cannot.

I did turn on the dynamic pose option, and that seems much better for me

As for the game, I just tried it on American Truck Simulator with freetrack (Vs simConnect) (didn't feel like buying Falcon BMS just to try this), and actually it works perfectly. Tracks my head in basically real time. So maybe this is a prepar3d setting i'm missing?

[image]
[image]

sthalik, 2020-03-31:
> So maybe this is a prepar3d setting i'm missing?

See if using EZCA helps. I'm ignorant about P3D and FSX but that's what FSX users generally do.

MathijsG, 2020-03-31:
Could this be related or of any help? https://github.com/opentrack/opentrack/issues/174 I always encounter a lot of trouble with those tons of subversions of simconnect. I don't know why Microsoft just didn't move to the more common trackir protocol every other game uses.

sthalik, 2020-03-31:
> why Microsoft just didn't move to the more common trackir protocol every other game uses.

I have to oppose this on principle :)

> Could this be related or of any help? #174

Only if your CPU cores are overloaded. But that's common for all output protocols.

eclair4151, 2020-03-31:
> See if using EZCA helps
id still rather not be buying software to test this out without knowing if it will do anything

and looking at that thread does not seem like the same issue. I guess there isnt really anything else i can do on your end. Looks like it must be a p3d issue or something. Maybe i will ask around on the flight sim subreddit and see if they have any ideas

eclair4151, 2020-03-31:
Is there a forum or subreddit community for opentrack i could reach out to as well?

eclair4151, 2020-04-02:
i couldn't solve the issues but using Xplane 11 with the X-Camera plugin works great so just gonna move to that. Thanks!

closed 2020-04-02 by eclair4151