opentrack/opentrack #1066: AI based tracker

gavazquez, 2020-03-23:
Hi. I made a tracker based on [OpenFace](https://github.com/TadasBaltrusaitis/OpenFace) and it uses artificial intelligence to track the head and orientation. This tracker is much more precise than the old "FaceApi" tracker that FaceTrackNoIR uses

I managed to send the parameters trough UDP to opentrack but I coudn't find any documentation about it.

Currently it "works" but the values are probably way too big or I may need to change the order of the values in the array that is sent trough UDP... Also, I only receive position (X,Y,Z) values but not the tilt angles.

Can you tell me what's the order of the parameters that must be sent trough UDP and what's the accepted range in them?

Here is how I send my data trough UDP:
https://github.com/gavazquez/GazeTracker/blob/master/GazeTracker/MainWindow.xaml.cs#L262

And I tried to look at this code of the opentrack but I'm still not sure of what I'm doing wrong as I don't know C++
https://github.com/opentrack/opentrack/blob/unstable/tracker-udp/ftnoir_tracker_udp.cpp

Here is the repo in case you want to have a look and try it, it should be easy to port to C++ aswell:
https://github.com/gavazquez/GazeTracker

bjensie, 2020-03-25:
I DL'ed it and as it doesn't seem to need any installation I just unpacked, ran the EXE and
got the message that it was searching for CAMs, then it crashed! Why? Do I need some
additional support files or what?

gavazquez, 2020-03-26:
There must be a problem with your USB cam. Anyway, try to put an issue in the gazetracker repo and we can follow it there. I'm closing this one

closed 2020-03-26 by gavazquez