opentrack/opentrack #1067: Cannot create or save profiles.

Tobytb, 2020-03-24:
I am running 2.3.11 and the tracking is working great however, I cannot create a new profile at all, when I enter a profile name and hit ok then check the folder no .ini is created.
I have tried downloading someone else's profile and it appears in opentrack but all the curves and settings remain as default and any changes I make are not saved.
I have checked permissions on the folder storing the .ini file and I have full control.
I have also tried previous versions and no luck, not sure if I am missing something in the setup 

Below shows screenshot with the downloaded profile shown but default.ini is always shown in the top left of the window.

[image]

Tobytb, 2020-03-24:
All ok, I got it working.
Had to copy the profile I download in the opentrack folder, rename it then I was able to edit and save both.

Tobytb, 2020-04-05:
Still having issue with profiles, I have removed open track and reinstalled and I still am unable to create a new profile (empty or copied) and any changes I make are not saved

sthalik, 2020-04-05:
Can you open the profile directory?

Tobytb, 2020-04-05:
I can open the profile directory, after initial install it is empty. 

I have tried creating a profile, after naming profile, returns to main window, no profile name and directory is still empty. 

I download a profile and added it to the directory manually which then appears in the profile drop down but any changes made are not saved (there is no save button so I assume this is done via the ok button?). 
If I try creating a copy of the manually added profile nothing is created (directory only has the manually added one). 
Thanks Toby

RCR-1909, 2020-04-10:
I have the same problem.  Cannot modify and save configuration files.  Have checked permissions and see no problems.  Default directory was not initially created.  Created it manually and the program does go to the directory.  Loaded the default.ini from one of the other versions and the program will use it, but not save any changes thereby requiring repeated changes prior to each use.

sthalik, 2020-04-18:
What happens if you use the portable version and create `portable.txt` in the directory with `opentrack.exe`? Does the config directory get created then?

Is there anything unusual going on with your configuration, i.e. Remote Desktop, home directory on a network share, or anything like that?

Tobytb, 2020-04-22:
I have tried the portable version (2.3.0 was the only one I could find) and this is working ok, I can create new profiles and the edits to profiles are saving.
My system is not unusual at all just running Windows 10 Pro 64bit with nothing fancy.
I will test more when I can.

RCR-1909, 2020-04-22:
Thanks for the tip.  Found 2.3 portable.  Will check back if/when I learn anything.

RCR-1909, 2020-05-12:
Finally got around to setting up and trying the portable version.  I still cannot get the program to setup or modify the default profile.

gibsonfm, 2020-05-18:
Hi guys , I have found the problem. The opentrack need the folder (..user\documents) to be stored on hard drive of the C:(system). In my case, I returned the "documents" to original place c:\users\...\documents (my documents is stored on another HD) , then after I set it up(opentrack profile curves) I back the folder to my another HD. I'm sorry for my English, let me know if I help you.

sthalik, 2020-05-18:
I always had my documents directory on the D drive and it worked just fine. At least it's working fine for you.

RCR-1909, 2020-05-18:
Odd.  I still cannot get it to save a new profile or modify the existing one, though the program "appears" to open to the proper directory within the program.  

I wonder if this is one of those anomalies in the transition to Windows 7 and earlier versions to 8 and above where Windows does some funny things with older apps and creates something like an invisible (to the user) link to a subdirectory in the C:\Users\*username*\AppData\Local that contains the data?  Some of the older programs that are expecting a certain behavior don't always play well with the later versions of Windows.  

None of this is seen by the user normally and you don't find out about it till you go about trying to figure out why something is not working.  I ran into this with some older engineering software that seemed to work fine, but I could not find its output files.  The program would pull them up fine from what appeared to be the correct directory, but that directory was nowhere to be found through normal browsing.  I had to unhide a series of directories in order to locate where Windows was "helping me" by putting them there.  It was something like an internal mklink command action.

I do not know if this is what is going on, but that is one theory.

RCR-1909, 2020-06-17:
Solved the problem.  

Windows, in its' infinite efforts to make things "cloud friendly" had started a "Onedrive Desktop" and everything visually on the desktop was actually in the Onedrive Desktop folder.

After copying the folders to the true local Windows desktop things started behaving correctly.

George174174, 2022-12-19:
Решил проблему. Версия 2.3.11
все параметры=> обновление и безопасность => безопасность windows => защита от вирусов и угроз => журнал защиты => "доступ к защищенным папкам заблокирован opentrack.exe" действия => разрешить на устройстве. 

all settings=> update and security => windows security => virus and threat protection => protection log => "access to protected folders is blocked opentrack.exe " actions => allow on device.

fabioduque, 2023-12-14:
> Решил проблему. Версия 2.3.11 все параметры=> обновление и безопасность => безопасность windows => защита от вирусов и угроз => журнал защиты => "доступ к защищенным папкам заблокирован [...]

This solved the issue for me! Opentrack was indeed blocked by Windows Security. After I allowed Opentrack, profiles started working again. Thanks

closed 2023-12-25 by sthalik