opentrack/opentrack #1071: Game Data numbers jumping - Accela issue?

Grimdude, 2020-04-11:
Hello,

Since yesterday (first observed) my headtracking gives a weird behavior despite no changes on my part since the previous session about 2 weeks ago.

From trying a number of things, it seems to me that the Accela filter is processing the Raw Data incorrectly.

Video of my issue: https://youtu.be/gin_AGJyuLY

I am using OpenTrack 2.3.11, a Delanclip IR kit and a modified PS3 Eye Camera.

I'd appreciate any help!

Thanks,
Grim

mathias-schwaller, 2020-04-13:
I have the same problem in the current version (2.3.11) with Aruco Input and PS3 Eye Camera.

Cryptoh, 2020-04-13:
I have the same problem... In the Point Tracker Settings the Camera FPS drops from 75 to ~5 and jumps back to 75 FPS

grgeiger, 2020-04-15:
I also experience this. I did just let java update but not sure that's related.

Grimdude, 2020-04-15:
Hello,

An update: after @grgeiger 's comment, I went to check if there were any Windows Updates that I had not been notified of (Windows 10 Pro).

There was, and I updated the following:
KB4537572 (.NET update)
KB4549951 (cumulative WIN update)

And lo and behold, headtracking is working as smoothly as ever once again!

I guess the .NET update was necessary. Apologies for not checking this before the post!

Should I keep the issue open for @mathias-schwaller and @Cryptoh to confirm if this fixes it for them before closing?

Thanks for everyone's input!
Grim

sthalik, 2020-04-18:
Note, opentrack doesn't use .NET in any form. This could be either the cumulative update, an undocumented part of the .NET update, or completely unrelated.

Please keep the issue open for now, and thank you for confirming the solution.

RazgrizXhunter, 2020-07-03:
Hi,
Had the exact same issue shown in the video today, although I'm using DIY EDTracker (and at first thought that the device was malfunctioning), couldn't get any filter working and updating windows was the solution (I don't really know how to check what was updated, but willing to give as much info to fix the problem as possible).
I don't know if I have to open a new thread to suggest this but, I didn't know I can choose not to use a filter until I saw @Grimdude 's video, maybe filling the blank space with something?

sthalik, 2022-09-30:
Can you confirm this with [2022.3.0](https://github.com/opentrack/opentrack/releases/tag/opentrack-2022.3.0)?

closed 2024-06-07 by sthalik