opentrack/opentrack #1072: View gets inverted at certain yaw value

ChrisTheAviator, 2020-04-12:
Hello there!

When I am using opentrack (2.3.11), I experience at a certain yaw value a 180 degree pitch, so when I was at first just looking to one side (it does not matter if left or right) and I pass a certain value, the view gets inverted. As for tracking model I use a selfmade cap model.

Now what I have noticed so far is that when the center (upper) LED and one of the outer LEDs build a vertical line with the center/upper LED, the view gets suddenly inverted, as the program would not be able to recognize the cap model anymore.

I have searched this and other forums about this issue but none of them seem to fit the exact issue as I have it here. As soon as im turning my head back to center a bit so the outer LED is again "outside" of the center/upper LED the octopus is again in normal view and just turning in the direction is it was before.

[image]

[image]

I have also read through an entire setup forum thread on how to configure the certain options and values which i striclty adhered to. 

I tried also freetrack again, which I have been using before, but since it had issues during startup and getting not updated anymore, stopped using it. There this certain behaviour is not appearing, as I can turn my head further aside until the camera loses one of the LEDs at some point, but without any flipping view.

sthalik, 2020-04-13:
Use the dynamic pose setting for caps.

closed 2020-04-13 by sthalik

ChrisTheAviator, 2020-04-13:
Yep that worked, thank you a lot! Now I feel really stupid, for overlooking this little line, everytime, I was checking through all the different menus.... *facepalm*