opentrack/opentrack #1074: binaries for Linux

efa, 2020-04-16:
step done with v.2.3.11:

```
$ mkdir build
$ cd build
$ cmake ..
...
CMake Error at CMakeLists.txt:33 (cmake_minimum_required):
  CMake 3.13 or higher is required.  You are running version 3.10.2

```
I installed last cmake, than run with that I got another error:
```
...
CMake Error at cmake/opentrack-qt.cmake:14 (find_package):
  Could not find a configuration file for package "Qt5" that is compatible
  with requested version "5.14.0".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake, version: 5.9.5
```

It is far too complicate for my skill to build on Linux.
Are there pre compiled binaries? DEB or AppImage are good to me.
thank you

sthalik, 2020-04-18:
> It is far too complicate for my skill to build on Linux.

That's the point of using Windows -- you don't have to deal with any of that.

> Are there pre compiled binaries? DEB or AppImage are good to me.

It's not feasible to provide binaries. `libc` isn't forward-compatible, and any shared library update will render the package useless. Finally, the Wine protocol won't work unless the wrapper was built with the exact same Wine version.

efa, 2020-04-18:
> That's the point of using Windows -- you don't have to deal with any of that.

With other packages I was able to build.
But the required dependences for this build are really updated. My distro (Ubuntu 18.04) has quite a previous version.

> It's not feasible to provide binaries. `libc` isn't forward-compatible, and any shared library update will render the package useless. Finally, the Wine protocol won't work unless the wrapper was built with the exact same Wine version.

this is why I cited also AppImage, these days many applications are distributed with this format (or other similar) and are all cross-distro:
https://github.com/AppImage/AppImageKit/wiki/Similar-projects

Take care that on Linux I want to test Opentrack together native X-Plane and FlightGear only, so I think I can avoid Wine dependencies

sthalik, 2020-04-18:
Hold on, are you building the stable release? The supported procedure is to build `unstable` on Linux. The Qt version requirement was dropped.

efa, 2020-04-18:
I tryed to build the release:
https://github.com/opentrack/opentrack/archive/opentrack-2.3.11.tar.gz
Should I try with current sources from git?

sthalik, 2020-04-18:
Yes.

efa, 2020-04-18:
OK, I had to install 'libprocps-dev', then cmake work.
Compilation ended with error:
```
...
[ 64%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o
/home/efa/c/opentrack/qxt-mini/qxtglobalshortcut_x11.cpp:40:10: fatal error: qpa/qplatformnativeinterface.h: File o directory non esistente
 #include <qpa/qplatformnativeinterface.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/build.make:128: recipe for target 'qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o' failed
make[2]: *** [qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o] Error 1

```

efa, 2020-04-18:
'qpa/' directory is completely missing in git sources

writequit, 2020-04-22:
Looks like you need to install qtbase5-private-dev

# apt-file search qplatformnativeinterface.h
qtbase5-private-dev: /usr/include/x86_64-linux-gnu/qt5/QtGui/5.12.5/QtGui/qpa/qplatformnativeinterface.h

I haven't tried compiling yet, just checking out my options.

efa, 2020-04-22:
OK, now build complete

efa, 2020-04-22:
the binary 'opentrack' is generated in:
build/opentrack
so to let start the application and found all its dependencies, I had to use a script like this:
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:../gui:../migration:../logic:../spline:../pose-widget:../api:../options:../compat:../qxt-mini; opentrack

The GUI start, Options and Mapping buttons works, but when I press Start I got an error dialog box that say:
Startup failure
Library load failure

and on the console is shown:
```
(opentrack:17996): GVFS-RemoteVolumeMonitor-WARNING **: 15:03:52.563: remote volume monitor with dbus name org.gtk.vfs.GoaVolumeMonitor is not supported
DEBUG [/home/efa/c/opentrack/logic/runtime-libraries.cpp:25]: protocol dylib load failure

```
maybe another library is needed?

sthalik, 2020-04-24:
No. Run `make install` first.

efa, 2020-04-24:
OK, I made the install, then selecting PointTracker and FlightGear, start open the webcam.
It show the image reversed upside-down, see attach
[image]
Other app like VLC show the image with the right orientation.

sthalik, 2020-04-27:
See if `v4l2-ctl -l -d /dev/video0` has any vertical flip options.

efa, 2020-04-27:
```
efa@02cor2130:~$ v4l2-ctl -l -d /dev/video0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=-10 value=-10
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=6500 value=6500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

```
where can be read the reference/coordinate top-left origin from the /dev/video0  device? Is there an API? Seems VLC can read the right orientation of the video stream

efa, 2020-04-27:
I think the head tracking cannot work with an image upside down, it can?

sthalik, 2020-04-27:
> I think the head tracking cannot work with an image upside down, it can?

Not really. I don't know what's causing this. For the moment use the following patch.

[patch-video-opencv-flip.diff.txt](https://github.com/opentrack/opentrack/files/4541511/patch-video-opencv-flip.diff.txt)

sthalik, 2020-04-27:
To help fix this problem, please locate the `libopencv2-core` library by executing the moral equivalent of:

```
# ldd $(dpkg -L libopencv2-core\* | xargs fgrep .so) | grep v4l
```

Basically I'm looking whether libopencv2-core.foo.so is linked against libv4l2 or the old version. There are bug reports of various software having that bug when using the legacy interface.

efa, 2020-04-27:
`efa@02cor2130:~$ apt-cache search libopencv
libopencv-apps-dev - Opencv_apps Robot OS package - development files
libopencv-apps0d - opencv_apps Robot OS package - runtime files
libopencv-calib3d-dev - development files for libopencv-calib3d3.2
libopencv-calib3d3.2 - computer vision Camera Calibration library
libopencv-contrib-dev - development files for libopencv-contrib3.2
libopencv-contrib3.2 - computer vision contrlib library
libopencv-core-dev - development files for libopencv-core3.2
libopencv-core3.2 - computer vision core library
libopencv-dev - development files for opencv
libopencv-features2d-dev - development files for libopencv-features2d3.2
libopencv-features2d3.2 - computer vision Feature Detection and Descriptor Extraction library
libopencv-flann-dev - development files for libopencv-flann3.2
libopencv-flann3.2 - computer vision Clustering and Search in Multi-Dimensional spaces library
libopencv-highgui-dev - development files for libopencv-highgui3.2
libopencv-highgui3.2 - computer vision High-level GUI and Media I/O library
libopencv-imgcodecs-dev - development files for libopencv-imgcodecs3.2
libopencv-imgcodecs3.2 - computer vision Image Codecs library
libopencv-imgproc-dev - development files for libopencv-imgproc3.2
libopencv-imgproc3.2 - computer vision Image Processing library
libopencv-ml-dev - development files for libopencv-ml3.2
libopencv-ml3.2 - computer vision Machine Learning library
libopencv-objdetect-dev - development files for libopencv-objdetect3.2
libopencv-objdetect3.2 - computer vision Object Detection library
libopencv-photo-dev - development files for libopencv-photo3.2
libopencv-photo3.2 - computer vision computational photography library
libopencv-shape-dev - development files for libopencv-shape3.2
libopencv-shape3.2 - computer vision shape descriptors and matchers library
libopencv-stitching-dev - development files for libopencv-stitching3.2
libopencv-stitching3.2 - computer vision image stitching library
libopencv-superres-dev - development files for libopencv-superres3.2
libopencv-superres3.2 - computer vision Super Resolution library
libopencv-ts-dev - development files for TS library of OpenCV (Open Computer Vision)
libopencv-video-dev - development files for libopencv-video3.2
libopencv-video3.2 - computer vision Video analysis library
libopencv-videoio-dev - development files for libopencv-videoio3.2
libopencv-videoio3.2 - computer vision Video I/O library
libopencv-videostab-dev - development files for libopencv-videostab3.2
libopencv-videostab3.2 - computer vision video stabilization library
libopencv-viz-dev - development files for libopencv-viz3.2
libopencv-viz3.2 - computer vision 3D data visualization library
libopencv3.2-java - Java bindings for the computer vision library
libopencv3.2-jni - Java jni library for the computer vision library
libopencv-ml2.4v5 - computer vision Machine Learning library
libopencv-calib3d2.4v5 - computer vision Camera Calibration library
libopencv-videostab2.4v5 - computer vision video stabilization library
libopencv-superres2.4v5 - computer vision Super Resolution library
libopencv-legacy2.4v5 - computer vision legacy library
libopencv-ts2.4v5 - computer vision ts library
libopencv-contrib2.4v5 - computer vision contrib library
libopencv-core2.4v5 - computer vision core library
libopencv-gpu2.4v5 - computer vision GPU library
libopencv-features2d2.4v5 - computer vision Feature Detection and Descriptor Extraction library
libopencv2.4-jni - Java jni library for the computer vision library
libopencv-ocl2.4v5 - computer vision OpenCL support library
libopencv-photo2.4v5 - computer vision computational photography library
libopencv-objdetect2.4v5 - computer vision Object Detection library
libopencv-flann2.4v5 - computer vision Clustering and Search in Multi-Dimensional spaces library
libopencv-video2.4v5 - computer vision Video analysis library
libopencv-imgproc2.4v5 - computer vision Image Processing library
libopencv-stitching2.4v5 - computer vision image stitching library
libopencv-highgui2.4v5 - computer vision High-level GUI and Media I/O library

efa@02cor2130:~$ apt-cache search videoio
libopencv-videoio-dev - development files for libopencv-videoio3.2
libopencv-videoio3.2 - computer vision Video I/O library

efa@02cor2130:~$ dpkg -L libopencv-videoio3.2 | grep -F .so
/usr/lib/x86_64-linux-gnu/libopencv_videoio.so.3.2.0
/usr/lib/x86_64-linux-gnu/libopencv_videoio.so.3.2
[... 213 lines cut ...]
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f2183725000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f21834e5000)
	libheimntlm.so.0 => /usr/lib/x86_64-linux-gnu/libheimntlm.so.0 (0x00007f21832dc000)
	libkrb5.so.26 => /usr/lib/x86_64-linux-gnu/libkrb5.so.26 (0x00007f218304f000)
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007f2182dad000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007f2182b77000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007f2182961000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007f2182738000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f2182529000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f21822df000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f21820a7000)`

Seems do not depend on v2l
To be sure I also searched in 'libopencv_video':

`efa@02cor2130:~$ dir /usr/lib/x86_64-linux-gnu/libopencv_video.so*
[... cut ...]
-rw-r--r-- 1 root root 346120 2018-09-13 20:10 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9
-rw-r--r-- 1 root root 403464 2018-09-20 11:28 /usr/lib/x86_64-linux-gnu/libopencv_video.so.3.2.0
lrwxrwxrwx 1 root root     24 2018-09-20 11:28 /usr/lib/x86_64-linux-gnu/libopencv_video.so.3.2 -> libopencv_video.so.3.2.0
lrwxrwxrwx 1 root root     22 2018-09-20 11:28 /usr/lib/x86_64-linux-gnu/libopencv_video.so -> libopencv_video.so.3.2

efa@02cor2130:~$ ldd /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9 | grep v4l
efa@02cor2130:~$ 
`

efa, 2020-04-27:
in the meantime I found a work around.
Made a script like the one below to run in background just before opentrack:

`#!/bin/bash
\# vertical flip of default webcam on video0

trap ctrl_c INT # connect function ctrl_c to INT
function ctrl_c() { # restore webcam to default/0
   sudo mv /dev/video0 /dev/video1
   sudo mv /dev/video2 /dev/video0 # restore webcam to default/0
   sudo rmmod v4l2loopback
}

\# sudo apt-get install v4l2loopback-utils
sudo modprobe v4l2loopback exclusive_caps=1 #create /dev/video1
sudo mv /dev/video0 /dev/video2 # webcam/default/0 become video2
sudo mv /dev/video1 /dev/video0
ffmpeg -f v4l2 -i /dev/video2 -vf "vflip" -f v4l2 /dev/video0 # webcam/video2 vertically flipped to video0
`

sthalik, 2020-04-27:
That was fixed in opencv 4.3.0. There's no bug to report. That patch should work for you till your distro upgrades, since I don't plan on touching that file.