opentrack/opentrack #1076: Issues with phone tracking

StanberyTrask, 2020-04-21:
I was recently able to set opentrack up with my phone but the data that the program was receiving seems nonsensical, with the octopus just flailing around seemingly with no connection to how my phone is moving or orientated. Nothing I have done with the filter settings or mapping seems to help and I have verified that it is in fact my phone that's sending the data through freepie with bitshark but im not able to make sense of what's being sent by the phone. I've linked the data that bitshark was picking up but its all raw hex so I have no idea if it's of any use.
[FreePie packet dump.txt…]()

sthalik, 2020-04-21:
Do you have all three of magnetometer, accelerometer and gyroscope on the phone?

StanberyTrask, 2020-04-21:
I should have said this before but im using an lg g8 and it has all 3. send raw data is also on as with it off the character wouldn't move at all as if I wasn't transmitting.

sthalik, 2020-04-21:
> send raw data is also on as with it off the character wouldn't move at all as if I wasn't transmitting.

Which FreePIE version are you using? There's one (or actually two) bundled with opentrack, and an older version bundled with FreePIE.

StanberyTrask, 2020-04-21:
oh im using the one that's linked here, https://github.com/opentrack/opentrack/wiki/Smartphone-Headtracking I guess that's an older version.
ok I got the freePIE from the contrib and it's working just fine, completely slipped my mind that there would be mismatched versions.

closed 2020-04-21 by StanberyTrask