opentrack/opentrack #1077: OpenTrack 2.3.11 + PS3 Eye Camera + DelanClip IR Not Recognized by Elite: Dangerous

SilverbackPrime, 2020-04-21:
__Issue__
After setting up OT, I am able to move the octopus as expected, with correct yaw, pitch, and roll. However, when I start Elite: Dangerous, my head movements are not reflected in game, nor can I set the up/down and left/right axes in the headlook control settings menu.

__OpenTrack Settings__
- Running as Administrator?: No
- Input: PointTracker 1.1 (All settings default except changing camera framerate to 75Hz)
- Output: freetrack 2.0 Enhanced (All settings default)
- Filter: Accela (All settings default)

__Elite: Dangerous Settings__
- Running as Administrator?: No

__Troubleshooting Attempts__
1. With OT installed on the same drive as Steam/Elite: Dangerous, and freetrack set to use both protocols - No change in ability to utilize headtracking in game
2. With OT installed on the same drive as Steam/Elite: Dangerous, and freetrack set to use only TrackIR protocol - No change in ability to utilize headtracking in game
3. With OT installed on the same drive as Steam/Elite: Dangerous, and freetrack set to use only freetrack protocol - No change in ability to utilize headtracking in game
4. With OT installed on C: (E: D installed on separate drive), and freetrack set to use both protocols - No change in ability to utilize headtracking in game
5. With OT installed on C: (E: D installed on separate drive), and freetrack set to use only TrackIR protocol - No change in ability to utilize headtracking in game
6. With OT installed on C: (E: D installed on separate drive), and freetrack set to use only freetrack protocol - No change in ability to utilize headtracking in game

sthalik, 2020-04-21:
Did you try installing in `c:/Program Files`?

SilverbackPrime, 2020-04-21:
attempts 4-6 were executed with OT installed in `c:/Program Files (x86)`
if it is an important distinction, I will reinstall to `c:/Program Files` and see if that corrects the issue

SilverbackPrime, 2020-04-21:
reinstalling to `C:/Program Files` and running both OT and ED as Administrator solved the issue.

I recommend updating the install/configure instructions to note the requirement to install in that directory, as the installer defaults to the (x86) folder due to OT being a 32-bit application.

sthalik, 2020-04-21:
This is entirely ED's fault. I'd prefer not to install in the incorrect (opentrack release is 32-bit) location because of a single developer's incompetence.