opentrack/opentrack #1078: Use your Face ID-enabled iPhone for 6 DoF tracking

yassineim, 2020-04-25:
Hi,

If you have a Mac with Xcode, and an iPhone with Face ID, you can use this app I just made:
https://github.com/yassineim/iphone-input-opentrack

This is just demo code from Apple, to which I added the UDP client logic.

To use the app:

1. build it with Xcode and run it on your iPhone
2. now in Windows, enable USB tethering (after installing iTunes)
3. in opentrack, choose "UDP over network" as an input method
4. that's it

This was tested with both DCS and ETS 2.

You may need to play with the invert checkboxes in opentrack.

Most of the relevant code is here: https://github.com/yassineim/iphone-input-opentrack/blob/master/ARKitFaceExample/Face%20Overlay%20Content/TransformVisualization.swift

Feel free to try it and ask any question.

tanc, 2020-05-15:
This is awesome! I just got it set up and working. For some reason I couldn't get USB tethering working in Windows 10 but it works just fine over wifi if you adjust the IP address on [line 39](https://github.com/yassineim/iphone-input-opentrack/blob/master/ARKitFaceExample/Face%20Overlay%20Content/TransformVisualization.swift#L39) of your code. It would be cool to have a setting for this and a simple form field to input an address but its easy enough to recreate the app with a new IP if needed. Thanks for this!

yassineim, 2020-05-15:
I’m so glad it worked for you. You’re right about the IP address, and cheers for finding that part.

So far the code needs 3 things:

1. Remove the code for the 3 other tabs (we don’t need them anymore)
2. Remove the tab bar
3. Add the IP field

I wouldn’t mind if someone did these tasks and published everything on the App Store. There is one of this kind already (https://apps.apple.com/us/app/kinotracker-pc-head-tracker/id1451351654) but it doesn’t work.

Enjoy :)