opentrack/opentrack #1079: kinect v2 pointtracker IR crash

voipmonitor, 2020-04-27:
Hi, 

I want to use the IR camera with my IR CAP but once I select the IR kinect camera and press start, the program crashes. Is the IR camera supposed to work in poin tracker?

sthalik, 2020-04-27:
Send me a minidump. See <<https://www.wintellect.com/how-to-capture-a-minidump-let-me-count-the-ways/>>, the Task Manager method.

voipmonitor, 2020-04-27:
it is 2.3.11 version and in fact the opentrack produced crash dump on its own which I have found in c:\Users\festr\AppData\Local\CrashDumps\opentrack.exe.3828.dmp   (attaching) 
[opentrack.exe.3828.zip](https://github.com/opentrack/opentrack/files/4541141/opentrack.exe.3828.zip)

Slion, 2020-08-29:
> Is the IR camera supposed to work in point tracker?
I'm pretty sure it isn't PointTracker expects a colour buffer if I recall well.
For Kinect use Easy Tracker instead.

closed 2020-08-29 by Slion