opentrack/opentrack #1080: Input trackers list seems to be missing some trackers

killkrt, 2020-04-30:
Hi,

As I understood with new release 2.3.11 the list of input trackers should include new trackers such as Kinect Face tracker, but I can see it in my list:

[image]

I've already installed the Kinect 1.8 SDK (I have a XBox 360 Kinect V1) and checked if file `Kinect20.Face.dll` is installed in my `modules` folder.

sthalik, 2020-04-30:
Launch opentrack from a Windows console window. You'll see debug messages relating to loading the trackers. I'd expect the Kinect tracker fails to load due to missing SDK. Verify the missing libraries with [Dependency Walker](https://www.dependencywalker.com/) on the module.

killkrt, 2020-05-01:
I've used a more modern tool, since Dependency Walker gave me a lot of _false negatives_.
And this is the output:
[image]
****

While Dependency Walker was reporting missing references to lot of: API-MS-WIN-* and EXT-MS-WIN-RTCORE-*, but as I read [here](https://stackoverflow.com/questions/36240215/dependency-walker-missing-dlls) that could be ignored in Windows 10.

Cardiako, 2021-01-09:
Hi, but im in the same problem, Im with kinect V1, not V2.
Im install SDK 1.8 and dont apear Traker for kinnect, before install SDK 2 (not valid for kienect V1) and now apear the tracker for kienct but dont work.

opentracker is not compatible with kinect V1???

Thanks and sorry for my english.

kaligot, 2021-02-12:
I have same problem with kinect v1 and kinect 360 , is opentrack compatible with kinect v1 really?

paegan69, 2024-01-20:
did anyone find a solution to this yet? I've had to use a workaround called MagicMau's Kinect Headtracker to send sensor data via udp to open track but have OT be able to read the data directly would be a lot more convenient.