opentrack/opentrack #1082: Documentation for Proton

DePingus, 2020-05-07:
Hi. I've read through your closed issues and see that opentrack supports Proton (many thanks to @jp7677 ). But you have zero documentation on how to do this. Is there any way you could add some basic docs to the wiki? "[Linux users should be software devs](https://github.com/opentrack/opentrack/wiki/common-issues-linux-osx)" is aging poorly.

I'm not a software developer, but I have 15+ years working with BSD and Linux. I've been at this for 8 hours today. I was able to get opentrack compiled with eigen3, opencv, libevdev, and wine support on Fedora 31. But no Proton games are working.

- libevdev - Using FreePIE everything looks fine in opentrack, evtest, and jstest-gtk. But Proton games (IL2 BoS and ED) only detect 1 random axis...only occasionally. Super weird. I see the axis moving in jstest, but nothing in game.
- wine - Using FreePIE again, games detect nothing at all.
- opencv - My (unmodded) PS3 cam works, but I haven't ordered an IR clip yet because I can't get any game to reliably detect input from opentrack. I would definitely upgrade from a cell phone on my head if I could get games to accept opentrack input.

I would post the opentrack logs, but there none. Or...are there logs? There are no docs for logs. I don't know.

sthalik, 2020-05-07:
> no Proton games are working.

For Proton to work, you need compatible `winegcc` due to `libwine` ABI. Consider building Proton on your own in order to produce the required development bits. If you're able to build Wine already, then [Proton's repository](https://github.com/ValveSoftware/Proton/) with the appropriate tag will work as-is. Please note that opentrack needs `winegcc` in `PATH` while you're building it.

DePingus, 2020-05-07:
I appreciate the quick response. I just looked at building proton and...I don't have another 18 hours in me tonight. I guess I'll look at this tomorrow. Thanks.

jp7677, 2020-05-07:
Hi @DePingus I'm on Fedora too, I didn't had to build Proton myself. The opentrack wine plugin built with system wine was fortunately compatible with stock Proton installation.

How far do you got? Where you able to start the wine plugin using wine from Proton (thus could you see the process in e.g. system monitor)?

wallcarpet40, 2020-05-07:
Fellow Linux user here, too. I'm using Opentrack+Trackhat Clip +PS3Eye cam (modded). Took me quite a while to get it working, since I didn't realize that I was supposed to use the Wine-plugin, instead of playing around with the vJoy. vJoy worked, but obviously didn't give me 6 degrees of freedom but instead worked... well, like a joystick. 

Also didn't figure out at first that installing opentrack or opentrack-git with my package manager (pacman or yay, since arch-based) didn't include the Wine-plugin by default. 

Finally after scrolling through all the closed issues in Github, I was able to build Opentrack by myself by doing: 

```
git clone https://github.com/opentrack/opentrack
cd opentrack
mkdir build
cd build
cmake ..
ccmake .
```

EIGEN3_DIR-NOTFOUND   <-- This is the only one that is not found, although installed on my system, but I have had Opentrack work without it.
.
.
 SDK_WINE: OFF      <-- Press space to turn Wine-plugin on.
```
press c to configure
press g to generate
make
make install
```

Now I have the Wine-plugin visible in Opentrack and have settings as follows:

https://i.imgur.com/Mx806U4.png

AppID 227300 is for Euro truck Simulator 2. I have tried other versions of Proton other than 5.0 in Opentrack and Steam.

I start tracking and then stop. I run Steam and launch Euro truck Simulator 2. Once game is running, I press Start on Opentrack.

This is the way it was working for me, until recently. I don't know what changed and and how to solve it, since I'm pretty much a noob, when it comes to Linux. 

I was beginning to have other problems with my system too and it escalated to a point that I decided to nuke the whole drive and install Linux again and start from scratch.

Now I'm at the point, where Opentrack builds fine, the squid moves and I see both Raw Tracker Data and Game Data change, but not working in ETS2.

Looking at the Task Manager, whenever I start tracking on Opentrack, I see:
`opentrack-wrapper-wine.exe.so`
`wineboot.exe --init`
`winedevice.exe`
`winedevice.exe`
`wineserver`

They stay on the screen for a few seconds and then they close. 
If I remember correctly, I used to see TrackIR.exe in that list, when it was working, but not anymore.

Sorry for the long text, but I would really want to get this working again.

I currently have installed:
Steam with Proton 5.0.7
wine 5.7-1

sthalik, 2020-05-07:
> I currently have installed:
> Steam with Proton 5.0.7
> wine 5.7-1

Is the wineserver version matching? Try launching the `.so` manually with Proton's `wine` executable.

wallcarpet40, 2020-05-07:
I guess the wineserver versions don't match?
Running `/usr/bin/wineserver -v` results in Wine 5.7
Running `/home/htpc/.local/share/Steam/steamapps/common/Proton\ 5.0/dist/bin/wineserver -v` results in Wine 5.0

Running:
`/usr/bin/wine /home/htpc/Downloads/opentrack/build/proto-wine/opentrack-wrapper-wine.exe.so`

```
002e:fixme:dwmapi:DwmIsCompositionEnabled 000000006DFA0434
0030:fixme:iphlpapi:NotifyIpInterfaceChange (family 0, callback 0x69ed306d, context 0xfd68e0, init_notify 0, handle 0x18cfa00): stub
002e:fixme:ntdll:NtQuerySystemInformation info_class SYSTEM_PERFORMANCE_INFORMATION
004a:fixme:dwmapi:DwmIsCompositionEnabled 6D95DD14
004c:fixme:iphlpapi:NotifyIpInterfaceChange (family 0, callback 0x6a0df537, context 0x10d9008, init_notify 0, handle 0x191fce8): stub
004a:fixme:ntdll:NtQuerySystemInformation info_class SYSTEM_PERFORMANCE_INFORMATION
wine: configuration in L"/home/htpc/.wine" has been updated.

```

Running:
`/home/htpc/.local/share/Steam/steamapps/common/Proton\ 5.0/dist/bin/wine /home/htpc/Downloads/opentrack/build/proto-wine/opentrack-wrapper-wine.exe.so
`
```
wineserver: using server-side synchronization.
0034:err:module:__wine_process_init L"C:\\windows\\system32\\winemenubuilder.exe" not found
002c:err:wineboot:process_run_key Error running cmd L"C:\\windows\\system32\\winemenubuilder.exe -a -r" (126).
006c:err:module:load_so_dll failed to load .so lib "/home/htpc/.local/share/Steam/steamapps/common/Proton 5.0/dist/bin/../lib64/wine/winebus.sys.so": libudev.so.0: cannot open shared object file: No such file or directory
006c:err:ntoskrnl:ZwLoadDriver failed to create driver L"\\Registry\\Machine\\System\\CurrentControlSet\\Services\\winebus": c0000142
003c:fixme:service:scmdatabase_autostart_services Auto-start service L"winebus" failed to start: 1114
0084:err:ntoskrnl:ZwLoadDriver failed to create driver L"\\Registry\\Machine\\System\\CurrentControlSet\\Services\\wineusb": c0000142
003c:fixme:service:scmdatabase_autostart_services Auto-start service L"wineusb" failed to start: 1114
wine: configuration in L"/home/htpc/.wine" has been updated.
00e0:err:module:__wine_process_init failed to load L"Z:\\home\\htpc\\Downloads\\opentrack\\build\\proto-wine\\opentrack-wrapper-wine.exe.so", error c000012f

```

Running the first wine command I can see the Wine-glass window appear and go away. The second command while running Steam's version of wine results only in those errors you see above.

sthalik, 2020-05-07:
That looks relevant.

```
libudev.so.0: cannot open shared object file: No such file or directory
```

jp7677, 2020-05-07:
@wallcarpet40 You also need to set `LD_LIBRARY_PATH` and friends when starting `opentrack-wrapper-wine.exe.so` manual. See https://github.com/opentrack/opentrack/blob/unstable/proto-wine/proton.cpp for all that should be set. The `libudev` is likely a result of that since Proton is built against the Steam Runtime and not  against your system libraries.

DePingus, 2020-05-07:
@jp7677 I really appreciate you taking the time to respond. I had some time to work on this again. Trying to build proton requires Vagrant which isn't playing nice with virt-manager for me. I gave that up after reading your comments.

So in trying to start the wrapper manually, I created a script based on one you had posted in another issue:

```
#!/bin/sh

PATH="/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/bin/:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/bin:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/usr/bin:/usr/lib64/qt5/bin:/usr/share/Modules/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin" \
WINEDLLPATH="/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/lib64//wine:/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/lib//wine" \
LD_LIBRARY_PATH="/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/lib64/:/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/lib/:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/pinned_libs_32:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/pinned_libs_64:/usr/lib64/atlas:/usr/lib64//bind9-export:/usr/lib64/llvm5.0/lib:/usr/lib64/iscsi:/usr/lib64/llvm6.0/lib:/usr/lib64/qt-3.3/lib:/usr/lib64/vtk:/usr/lib/wine:/usr/lib64/wine:/lib:/lib64:/lib/sse2:/lib64/sse2:/lib64/tls:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/i386/lib/i386-linux-gnu:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/i386/lib:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/i386/usr/lib/i386-linux-gnu:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/i386/usr/lib:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/lib/x86_64-linux-gnu:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/lib:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/usr/lib/x86_64-linux-gnu:/home/depingus/.local/share/Steam/ubuntu12_32/steam-runtime/amd64/usr/lib" \
WINEPREFIX="/home/depingus/.local/share/Steam/steamapps/compatdata/307960/pfx" \
WINEESYNC=1 \
WINEFSYNC=0 \
OTR_WINE_PROTO=1 \

"/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/bin/wine" /home/depingus/repos/opentrack/build/proto-wine/opentrack-wrapper-wine.exe.so
```
This produced the same libudev.so.0 error @wallcarpet40 is getting. I have the same winserver versions he has too; 5.0 in proton and 5.7 in system. And I just guessed at OTR_WINE_PROTO. I don't know what goes there.

I looked up libudev.so.0 and found I could easily link the existing libudev.so.1 like this:
```
sudo ln -s /usr/lib64/libudev.so.1 /usr/lib64/libudev.so.0
```
Unfortunately, that still doesn't help me start the wrapper. I get:
```
wineserver: using server-side synchronization.
0034:err:module:__wine_process_init L"C:\\windows\\system32\\winemenubuilder.exe" not found
002c:err:wineboot:process_run_key Error running cmd L"C:\\windows\\system32\\winemenubuilder.exe -a -r" (126).
006c:err:ntoskrnl:ZwLoadDriver failed to create driver L"\\Registry\\Machine\\System\\CurrentControlSet\\Services\\wineusb": c0000142
003c:fixme:service:scmdatabase_autostart_services Auto-start service L"wineusb" failed to start: 1114
007c:fixme:plugplay:build_report_descriptor Clamping joystick to 6 axis
00a4:err:module:__wine_process_init failed to load L"Z:\\home\\depingus\\repos\\opentrack\\build\\proto-wine\\opentrack-wrapper-wine.exe.so", error c000012f
```
I'm a little confused on the order I'm supposed to start things in. Let me see if I have this straight:

- Launch game.
- Launch /usr/local/opentrack/bin/opentrack
- Start tracking in opentrack app
- Launch opentrack-wrapper-wine.exe.so with "/home/depingus/.local/share/Steam/steamapps/common/Proton 5.0/dist/bin/wine"

Do I need to launch /usr/local/opentrack/bin/opentrack with the same env variables as the wrapper? Did I guess OTR_WINE_PROTO correctly?

jp7677, 2020-05-07:
@DePingus Regarding your last question, actually opentrack should take care of starting opentrack-wrapper-wine.exe.so with the correct environment, so no need to do anything manually. When you start tracking, do you see a running process for opentrack-wrapper-wine.exe.so?

I'm seeing the libudev.so.0 error only when I don't set `LD_LIBRARY_PATH`. Just to be sure, could you remove that empty line from your script?

wallcarpet40, 2020-05-07:
I've been getting cross-eyed trying to figure this thing out and it's beginning to get too complicated for me to understand. 

As one final desperate act I read on the WineHQ FAQ, how to build wine from source. I managed to build an earlier version of wine (5.0), so the versions would match with Proton and it worked! 

But then I remembered an AUR-program called `downgrade` which I used to downgrade wine to version 5.6 and then I built Opentrack once again and now everything is working again!

I can't tell what is different with 5.6 and 5.7 of wine, but for me, this is good enough.

DePingus, 2020-05-07:
@jp7677 You were right about the extra lines in my bash script. Most of my errors disappeared when I removed them. I was also able to remove that sym link I had made. No new process for the opentrack wrapper ever opens though. I still get one last error:
```
0228:err:module:__wine_process_init failed to load L"Z:\\home\\depingus\\repos\\opentrack\\build\\proto-wine\\opentrack-wrapper-wine.exe.so", error c000012f
```
I'm going to try downgrading my system wine as @wallcarpet40 suggested to see if that helps. Thanks again!

DePingus, 2020-05-08:
Alright let me just get it out of the way. I'm an idiot. I thought wine-devel in Fedora's repo was the development files for wine. Turns out it is the development _branch_ (5.7). I needed wine-stable and wine-stable-devel to get 5.0.

After sorting that, I was able to successfully compile. This time, when I start tracking from opentrack, I see opentrack-wrapper-wine.exe.so in my running programs. It stays there until I stopped tracking from the app.

But games still don't detect any input. No matter how much I make that tiny octopus dance. I tried all the ESYNC/FSYNC combinations. And freetrack/npclient/both. I switched over to testing with Elite Dangerous (since @jp7677 confirmed it working with freetrack). Wiped its prefix directory. Ran fresh protontricks. Nothing.

Proton logs show some opentrack stuff going on:
```
49094.667:0314:031c:trace:loaddll:load_native_dll Loaded L"Z:\\home\\depingus\\repos\\opentrack\\build\\install\\libexec\\opentrack\\FreeTrackClient64.dll" at 0x61e80000: native
49190.831:0314:031c:trace:loaddll:free_modref Unloaded module L"Z:\\home\\depingus\\repos\\opentrack\\build\\install\\libexec\\opentrack\\FreeTrackClient64.dll" : native
```
And checking the registry for ED turned up a Path entry for FreetrackClient under HKCU:Software\Freetrack
```
Z:/home/depingus/repos/opentrack/build/install/bin/../libexec/opentrack/
```
I noticed the proton wineserver binary is 64bit. So I removed wine-stable and wine-stable-devel, then installed wine-stable64 and wine-stable64-devel and recompilied. I did **everything** all over again. Same exact results; down to the proton log and registry.

I'm out of ideas.

jp7677, 2020-05-08:
@wallcarpet40 @DePingus Thanks a lot for the research. So, as @sthalik suggested, it seems that Wine-staging 5.7 is no longer ABI compatible with Proton 5.0 (which is based on Wine 5.0). So using wine-stable 5.0 is likely the safest way to compile opentrack against. Wine-staging <= 5.6 seems to be OK too. 
Actually `error c000012f` yields `Bad Image`, so this more or less confirms it.

jp7677, 2020-05-08:
@DePingus I think you are really close. Could you try the following:
1) Start Elite Dangerous from Steam. Wait until the ED launcher appears.
2) Start opentrack. Verify opentrack Proton settings (Steam ID, etc). If you haven't changed anything with E-SYNC/F-F-SYNC in Steam, keep both options enabled. Use `Freetrack` or `Both` protocols.
3) Start tracking and verify that the wrapper process is started and keeps running. Check that opentrack reacts to your input.
4) Now start ED from the ED launcher and enter game play. It should "just work".

DePingus, 2020-05-08:
Thanks for clearing up the launch procedure. That's how I'm doing it. I verified the Steam ID for Elite Dangerous (359320). The proton log says `esync: up and running` but makes no mention of `fsync`. I tried all the possible combinations there. I also tried `both` protocols and just `freetrack`.

When I start tracking in opentrack I see `opentrack-wrapper-wine.exe.so` in my process monitor. It stays running until I press stop. The little pink octopus dances his heart away when I move my phone (using FreePIE UDP). I also tried with the built in test tracker. Both the raw tracker data and game data register changes. It just doesn't get detected in game. My regular HOTAS works.

I've had this install of Fedora since version 28. I think I may just wipe the whole thing and start fresh with v32. Then start again. This thing has turned into "Software Developer Simulator 2020". Thanks again!

jp7677, 2020-05-08:
Mmh, everything sounds good. The proton logs confirmed that
freetrackclient64 had been loaded? Do you have by chance anything enabled
for mouse head look in ED? I’ve found an article mentioning that this might
conflict.

Your system should be fine though, I’m also on Fedora 32 by upgrading
several times from earlier versions.

DePingus, 2020-05-09:
ED was freshly installed just to test this. The registry also showed a `NatrualPoint` entry after I enabled `both` protocols in opentrack. The only weird thing I did (over a year ago) was [increase the system's open file limits](https://www.reddit.com/r/SteamPlay/comments/9kqisk/tip_for_those_using_proton_no_esync1/) so that I did _not_ have to use `PROTON_NO_ESYNC=1` as a launch option in some games.

I'm trying to make some time to do a fresh install this weekend. I'll post here again when I get it done and I retry this mess. Thanks again for all the help. I really appreciate you taking the time.

sthalik, 2020-05-09:
See #1077, #977, and *common issues* on wiki.

sthalik, 2020-05-09:
> I think I may just wipe the whole thing and start fresh with v32. Then start again. This thing has turned into "Software Developer Simulator 2020". Thanks again!

Which is why I'd only recommend Linux for developers or admins.

jp7677, 2020-05-09:
To verify what I wrote above I did a fresh build on my upgraded Fedora 32 installation. This is what I did:

**Ensure that qt5-devel and wine-devel (both i686 and x86_64) are installed**

Note that I did not setup any other repositories. I didn't used winehq repositories, just the defaults from Fedora 32.

```bash
dnf install qt5-devel wine-devel.i686 wine-devel.x86_64
```
This currently gives me Wine-Staging 5.7. 

**Repeat the step mentioned above**

```bash
git clone https://github.com/opentrack/opentrack
cd opentrack
mkdir build
cd build
cmake ..
ccmake .
```
- Change installation prefix to `/home/user/bin` 
- Enable the wine plugin: SDK_WINE: OFF <-- Press space to turn Wine-plugin on.
- press c to configure
- press g to generate

```bash
make
```

Doing `make` gave me a compile error:

```bash
[ 80%] Generating opentrack-wrapper-wine.exe.so
In file included from /usr/include/wine/windows/windows.h:64,
                 from /home/user/Develop/opentrack/compat/shm.h:10,
                 from /home/user/Develop/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:16:
/usr/include/wine/windows/winsock.h:445:16: error: redefinition of ‘struct timeval’
  445 | typedef struct WS(timeval)
      |                ^~
In file included from /usr/include/bits/timex.h:22,
                 from /usr/include/bits/time.h:73,
                 from /usr/include/time.h:33,
                 from /usr/include/pthread.h:23,
                 from /usr/include/c++/10/x86_64-redhat-linux/32/bits/gthr-default.h:35,
                 from /usr/include/c++/10/x86_64-redhat-linux/32/bits/gthr.h:148,
                 from /usr/include/c++/10/ext/atomicity.h:35,
                 from /usr/include/c++/10/memory:75,
                 from /home/user/Develop/opentrack/proto-wine/wine-shm.h:18,
                 from /home/user/Develop/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:8:
/usr/include/bits/types/struct_timeval.h:8:8: note: previous definition of ‘struct timeval’
    8 | struct timeval
      |        ^~~~~~~
winegcc: /usr/bin/g++ failed
``` 
I could fix the compile error by changing the order of includes in `proto-wine/opentrack-wrapper-wine-main.cxx`. The DXVK issue tracker hold a similar issue, this seems to be the result of newer wine versions. I'm not a c/c++ programmer, so really not sure if this is the correct way of doing things. What I changed was:

```diff
diff --git a/proto-wine/opentrack-wrapper-wine-main.cxx b/proto-wine/opentrack-wrapper-wine-main.cxx
index 3df3abc7..9248a212 100644
--- a/proto-wine/opentrack-wrapper-wine-main.cxx
+++ b/proto-wine/opentrack-wrapper-wine-main.cxx
@@ -5,7 +5,6 @@
 #include <cstdio>
 
 #include "freetrackclient/fttypes.h"
-#include "wine-shm.h"
 #include "compat/export.hpp"
 
 enum Axis {
@@ -14,6 +13,7 @@ enum Axis {
 
 #define __WINE_OLE2_H
 #include "compat/shm.h"
+#include "wine-shm.h"
 
 void create_registry_key(void);
 
@@ -42,7 +42,6 @@ private:
     void* mem;
     void *hMutex, *hMapFile;
 };
-#include <windows.h>
 
 int main(void)
 {
```

Building and installing succeeded now:

```bash
make
make install
```
**Start tracking for a Proton game, first attempt**

Though start tracking for a Proton game failed. The wrapper process came up, but terminated instantly. Likely with the above mentioned `Bad Format` error. I didn't run the wrapper manually to verify. (Note to self, do this!).

**Downgrade Wine**

What I did now was downgrading my wine version, based on the findings above. The Fedora 32 repository fortunately also contains Wine staging 5.5. So I picked this one:

```bash
dnf install wine-devel-5.5-1.fc32.i686 wine-devel-5.5-1.fc32.x86_64
``` 
Now I repeated the build step:

```bash
make clean
make
make install
```

The include modifications where still needed for Wine 5.5. 

**Start tracking for a Proton game, second attempt**

Start tracking for a Proton game works now here, the process stays open until tracking is stopped.
Testing with Elite Dangerous also succeeds here.  

- Start Elite Dangerous from Steam. Wait until the ED launcher appears.
- Start `opentrack`. Set the ED Steam ID, I kept E-SYNC/F-F-SYNC enabled and used the Freetrack protocol.
- Start tracking,  the wrapper process is started and runs.
- Start ED from the ED launcher and enter solo game play. Moving my Phone results in moving the head in ED.

Note that I haven't change my Steam installation, so I'm just using my home directory as library  location.

Since I'm running opentrack only against Proton I can safely upgrade or even remove Wine from my system. Well, at least until I start a new attempt to compile opentrack ;)

I hope this (more or less) brain dump helps a bit.
(@sthalik, assuming the header modification are correct, if you want I can create a PR for this.)

sthalik, 2020-05-09:
> @sthalik, assuming the header modification are correct, if you want I can create a PR for this.

That shouldn't be happening. As a workaround, try adding `#define NOWINSOCK` before the include.

jp7677, 2020-05-09:
> As a workaround, try adding `#define NOWINSOCK` before the include.

I've tried adding `#define NOWINSOCK` before `#include "compat/shm.h"` in `opentrack-wrapper-wine-main.cxx`, but that gives me the same error.

DePingus, 2020-05-10:
Success! I got it working in Elite Dangerous!

Turns out, I'm a bigger idiot that we previous thought. It was probably working before my last post. I've never used any tracking before. I thought I had to map the tracker to view controls in the game. I didn't realize until @jp7677 posted his last instructions that it "just works" with no mapping. All my "testing" was done in the settings for the game's control mapping and not in the actual cockpit. :facepalm:

Anyways, I ended up not wiping my whole system. I took @sthalik comment as a personal challenge! I did an in place upgrade to Fedora 32 (which even involved some lvm stuff with a 2nd drive). I did wipe all my games, steam, wine, and all their settings before starting on opentrack again.

I used @jp7677's latest set of instructions. I had the same results. Wine 5.7 didn't work. I downgraded to `wine-devel-5.5-1.fc32.i686` and `wine-devel-5.5-1.fc32.x86_64` and compiled with those. (Side note: it seems Fedora 32's repos got rid of the `wine-stable-devel` nonsense I dealt with in F31.) I can also confirm Wine 5.5 gave the same `struct timeval` compile error @jp7677 got. His edit to `proto-wine/opentrack-wrapper-wine-main.cxx` solved that.

After I successfully compiled and installed, running opentrack gave me a new error in the terminal:
```
QSocketNotifier: Can only be used with threads started with QThread
```
Opentrack launched, but it had a different color title bar, nothing in the output drop down, and only PointTracker in the input drop down. I was able to fix that by removing the `qt5-qtwayland` package with dnf. With that resolved, opentrack's gui behaved normally and I also got the plugin running when I started tracking.

Anyways, It's definitely working in Elite Dangerous now! Unfortunately, I still haven't gotten it working in IL-2 Sturmovik: Battle of Stalingrad (the game that got me started down this path). But I'm still messing around with it.

Thanks again to everyone who helped get this information out. As Proton gets better, more non software devs are bound to find their way to Linux. I hope this helps other people. I know it helped me!

wleoncio, 2020-08-01:
> Now I have the Wine-plugin visible in Opentrack and have settings as follows:
> 
> https://i.imgur.com/Mx806U4.png
> 

@wallcarpet40, how does one get to that window? I can't find it anywhere on my opentrack installation. :-/

wallcarpet40, 2020-08-01:
> @wallcarpet40, how does one get to that window? I can't find it anywhere on my opentrack installation. :-/

I have circled the button on this image: https://imgur.com/Q7SxOLq

wleoncio, 2020-08-01:
> https://imgur.com/Q7SxOLq

Interesting. When I click there, I just get a dialog message saying "there are no settings necessary for the Wine protocol". Maybe there's something missing on my end; this might also be related to issue #1120 I just opened.

wallcarpet40, 2020-08-01:
Probably some dependency missing or something went wrong with the compilation, I would guess. (not an expert though)

Steps to make it work on my computer (Arch-based linux):
`Wine` version has to be lower than `5.7 `(5.6 works for me), before compiling Opentrack.

```
git clone https://github.com/opentrack/opentrack
cd opentrack
mkdir build
cd build
cmake ..
ccmake .
```

During `ccmake .` my screen looks like this: https://imgur.com/ocPzdwZ

EIGEN3_DIR-NOTFOUND <-- This is the only one that is not found, although installed on my system, but I have had Opentrack work without it. If you see any other DIR-NOTFOUND, you need to install those dependencies first.

.
.
`SDK_WINE: OFF `<-- Press space to turn Wine-plugin on.

```
press c to configure
press g to generate
make
make install
```

wleoncio, 2020-08-01:
Thank you for the help, @wallcarpet40! After a more careful inspection, I noticed that instead of `SDK_WINE` I only have `SDK_WINE_NO_WRAPPER` here, which I suppose should have been left at `OFF`. Now I'm stuck at a failing `Gerenating opentrack-wrapper-wine.exe.so`, but I'll try to figure this out on my own and open a different ticket if it's the case (I may have highjacked too much of this thread already).

LaurentBouchard, 2020-08-03:
Thanks to all of you, I've been able to successfully make it work with E:D. Here's some notes that might help you in the future:
- The game **must** be located in `~/.local/share/Steam/steamapp`, otherwise opentrack will not be able to find it.
- On Archlinux, you **must** downgrade  `wine` to a version lower than 5.7. Using the [Arch Linux Archive](https://archive.archlinux.org/packages/w/wine/), you can get a package to allow the downgrade. Then, `sudo pacman -U wine-5.6-1-x86_64.pkg.tar.zst` should do the trick.
- For E:D, you need to invert the pitch axis.

My opentrack wine config: https://i.imgur.com/Q4STGv0.png

Other than that, just follow @jp7677 's instructions

sthalik, 2020-08-08:
> On Archlinux, you must downgrade wine to a version lower than 5.7

And why is that?

wallcarpet40, 2020-08-08:
@sthalik After some testing I've found out that: 

if `wine` version is **<**  `5.7` then I can build the `proto-wine` module in Opentrack and use the current latest **official** Proton from Valve, which currently is `5.0-9`

if `wine` version is **>=** `5.7` then I can build the `proto-wine` module in Opentrack and use some versions of Proton that are based on later builds of wine, such as GloriousEggroll's latest stable build, which currently is `5.9_GE_5_ST-2`

But I can't mismatch those. `proto-wine` module built with `wine 5.6 `cannot use any later Proton version than `5.0-9`
The same vice versa. `proto-wine` module built with `wine 5.7` or later cannot use Proton `5.0-9` or older.

waterlubber, 2020-09-06:
> @sthalik After some testing I've found out that:
>
> if `wine` version is **<** `5.7` then I can build the `proto-wine` module in Opentrack and use the current latest **official** Proton from Valve, [...]

How did you get OpenTrack to detect Proton-GE? I installed it from the AUR, steam detects it and it works, but Opentrack doesn't show it.

Note that I have two steam directories, one on a different drive, but I don't think that should make too much of a difference

miIktoast, 2021-01-14:
@DePingus 
> Anyways, It's definitely working in Elite Dangerous now! Unfortunately, I still haven't gotten it working in IL-2 Sturmovik: Battle of Stalingrad 

have you managed to get it working in il2 yet? Ive reread this thread a million times and tried building it with newer and older versions of wine/proton. currently arch-based on wine 5.9, proton 5.13-4, the wine wrapper is runs and opentrack detects il2 and autostarts its profile, I am launching opentrack after I start il2, doesnt even work in elite for me though most of my "testing" has been in il2

wallcarpet40, 2021-01-14:
@miIktoast After you built Opentrack, did you `sudo make install` or did you `make install`?
`sudo make install` will install Opentrack to `/usr` directory and I have noticed that I can't get Opentrack to work anymore.
But when I do `make install`, it will install Opentrack to the build-directory under `install/bin`. And then it works with my games (I'm testing with Euro Truck Simulator 2).

I have noticed this behavior on Arch-based linux and also OpenSUSE Tumbleweed, so I think that it has something to do with Steam and it's pressure vessel-container.

DePingus, 2021-01-14:
> have you managed to get it working in il2 yet? Ive reread this thread a million times and tried building it with newer and older versions of wine/proton. currently arch-based on wine 5.9, proton [...]

@milktoast

No, I never did get it working in IL-2:BoS. I got back into Elite for a little bit since it was working there and then ended up going back to Windows for MSFS2020.

When I was "testing" in Elite I was trying to map my views in the control menus but wasn't seeing any new axis/controllers to map to. Turns out I was way off. IIRC, you don't actually map it to any controls. It just works when you're in the cockpit.

miIktoast, 2021-01-14:
@wallcarpet40 > @miIktoast After you built Opentrack, did you `sudo make install` or did you `make install`?
I didn’t do ‘sudo make install’ though ‘opentrack’ folder is currently sitting in my downloads folder, will try moving to install/bin and see if that makes any difference

miIktoast, 2021-01-14:
@wallcarpet40 just to preface I'm new to linux so I may be doing something wrong but I believe I have all the dependencies. so I recompiled again just to make sure I didn mess anything up, built with `wine-5.9` running with `proton 5.13-4` which if 
>if wine version is < 5.7 then I can build the proto-wine module in Opentrack and use the current latest official Proton from Valve, which currently is 5.0-9

is still true, it should work. opentrack is placed in `/home/milktoast/opentrack` with it being built into `/home/milktoast/opentrack/build` where `make install` output all the subdirectories etc. I tried Euro Truck Simulator 2 (though proton), Elite: Dangerous and IL-2 Sturmovik: Battle of Stalingrad but have not had any luck. Opentrack still has no problem detecting the game, though the game doesnt detect opentrack input

wallcarpet40, 2021-01-14:
Could you make sure that you have the same settings as I do (I mean for Euro Truck Simulator 2)

Esync and Fsync enabled, correct Steam application id (227300 for ETS2), same Proton version selected as in Steam.

Also for ETS2, there is a config-file that I had to change (**config.cfg**) and it's located in folder:
`~/.local/share/Steam/steamapps/compatdata/227300/pfx/drive_c/users/steamuser/My Documents/Euro Truck Simulator 2`

There is a line that says `uset g_trackir "0"`
Change that to `uset g_trackir "1"`

Here are my settings for Opentrack:
[https://i.imgur.com/WEwat33.png](url)

DePingus, 2021-01-15:
@miIktoast Hey just thought of something. I remember back on Fedora before messing with OpenTrack, I had to add my user to the "input" group in order to get my (Steam) controller working. Maybe that has something to do with it?

miIktoast, 2021-01-15:
@wallcarpet40 Success! well... almost, I dont believe I changed anything other than the filter but Elite and Euro Truck are working now, though IL2 is not and it is the one I need opentrack to work with.

wallcarpet40, 2021-01-15:
@miIktoast That's good news! At least you know, it can work. Maybe there is a config-file in IL2 that enables TrackIR or somewhere in the game-settings. I've never played that game, so I wouldn't know.

miIktoast, 2021-01-16:
@jp7677  sorry to tag you in this put since youve done work on the proton output I thought you may be able to help. as my previous comments Ive been unable to get headtracking to work in IL-2 Sturmovik, I believe it has somehting to do with the freetrack output as it is not being loaded.
IL-2 Sturmovik
`1537.271:00d8:00dc:trace:loaddll:build_module Loaded L"Z:\\home\\milktoast\\opentrack\\build\\install\\libexec\\opentrack\\NPClient64.dll" at 000000006A940000: native`
Elite: Dangerous
`2028.939:02b8:02c8:trace:loaddll:build_module Loaded L"Z:\\home\\milktoast\\opentrack\\build\\install\\libexec\\opentrack\\NPClient64.dll" at 00000000614D0000: native
2028.940:02b8:02c8:trace:loaddll:build_module Loaded L"Z:\\home\\milktoast\\opentrack\\build\\install\\libexec\\opentrack\\FreeTrackClient64.dll" at 00000000614E0000: native`

jp7677, 2021-01-17:
HI @miIktoast no worries for tagging me. I’ve never tested opentrack with IL2 but from what I read here everything looks correct. It is fine that IL2 doesn’t load ‘freetrack.dll’ since it already does load ‘npclient64.dll’. It should always be one or the other since those are two competing implementations for head tracking (with ED being the exception here which tries both). Since opentrack works in other games for you, I would guess that your setup is correct and it is something in-game.
If you really want to verify your setup, if you own ACC (Assetto Corsa Competizione), you might wanna try this one since I remember this game worked with opentrack using npclient64.dll only.

miIktoast, 2021-01-17:
@jp7677 acc does indeed work. Is there a potential fix or am I sol until it starts working after an update some day?

There are no head tracking options in il2 that I’m aware of other than an ‘oldtrakir’ config option, if I enable it ‘NPClient’ doesn’t appear in the logs (though it could’ve been due to me being tired and missing it). Il2 worked with opentrack on windows with the ‘freetrack’ output  so I have no doubt whether the game is compatible with head tracking

jp7677, 2021-01-17:
@miIktoast I'm sorry, but no clue what could be wrong with IL2. Reading that IL2 accepts freetrack on Windows is interesting, considering that in loads NPClient on Linux. Though no idea where to start looking except consulting wine logs which you already did.

miIktoast, 2021-01-17:
@jp7677 here are my logs, I'm not sure what else I could be looking for, all I did was `ctrl+f` and search for opentrack. https://files.catbox.moe/325lcy.log

jp7677, 2021-01-18:
@miIktoast I'm sorry, but there isn't anything obvious that comes to my mind. The dll is loaded but dunno what IL2 is doing with it. You could try to build a debug version of  https://github.com/opentrack/opentrack/blob/master/contrib/npclient/npclient.c and see if the generated logs reveal something, though that needs quite some work.

jfwells, 2021-02-16:
Hi @jp7677 et al, I'm having problems getting opentrack-wine-wrapper.exe.so to load successfully in a steam prefix that is already running. Specifically, Proton experimental 5.13 running msfs2020. (on Arch btw).

If I start tracking before the game is running, the wrapper is spawned successfully.

Using a system 32-bit wine prefix, I can also get client.exe to work and display tracking output, so I know it can run.

Assuming a winegcc ABI mismatch as above, I've tried compiling with various wine versions, including 5.13, and older versions as suggested above, but the behaviour is alwaysthe same.

I spammed several qDebugs in `ftnoir_protocol_wine.cpp` and the QProcess wrapper StdError is giving `"03d4:err:module:__wine_process_init failed to load L\"Z:\\\\usr\\\\libexec\\\\opentrack\\\\opentrack-wrapper-wine.exe.so\", error c000000f\n"`

Recreating the environment and trying to load the module manually, I get a little more information:
````bash
$ WINEDEBUG=warn+all WINEESYNC=1 WINEFSYNC=1 PATH=xxx OTR_WINE_PROTO=3 LD_LIBRARY_PATH=xxx WINEPREFIX=xx /path/to/Proton\ -\ Experimental/dist/bin/wine64 /usr/libexec/opentrack/opentrack-wrapper-wine.exe.so

042c:err:heap:HEAP_GetPtr Invalid heap 0000000000000000!
042c:warn:file:NtCreateFile L"\\??\\Z:\\usr\\libexec\\opentrack\\kernelbase.dll" not found (c0000034)
042c:warn:module:load_dll Failed to load module L"Z:\\usr\\libexec\\opentrack\\opentrack-wrapper-wine.exe.so"; status=c000000f
042c:err:module:__wine_process_init failed to load L"Z:\\usr\\libexec\\opentrack\\opentrack-wrapper-wine.exe.so", error c000000f
````

I assume the kernelbase line is irrelevant?? Any ideas what might be going wrong?

sthalik, 2021-02-16:
Run `ldd` and `LD_DEBUG=bindings,files wine opentrack-wrapper-wine.exe.so`.

jfwells, 2021-02-16:
Hi @sthalik, thank you very much for the response.

````bash
$ ldd /usr/libexec/opentrack/opentrack-wrapper-wine.exe.so

        linux-gate.so.1 (0xf7fc0000)
        librt.so.1 => /usr/lib32/librt.so.1 (0xf7f5b000)
        libstdc++.so.6 => /usr/lib32/libstdc++.so.6 (0xf7d8a000)
        libm.so.6 => /usr/lib32/libm.so.6 (0xf7cbc000)
        libc.so.6 => /usr/lib32/libc.so.6 (0xf7ac4000)
        libgcc_s.so.1 => /usr/lib32/libgcc_s.so.1 (0xf7aa6000)
        libpthread.so.0 => /usr/lib32/libpthread.so.0 (0xf7a84000)
        /usr/lib/ld-linux.so.2 (0xf7fc2000)
````

(initially `ldd` also reported that I didn't have executable permission for `opentrack-wrapper-wine.exe.so` , but behaviour is still the same after making it executable)

````bash
$ LD_DEBUG=bindings,files wine /usr/libexec/opentrack/opentrack-wrapper-wine.exe.so &>debug.txt
````
[debug.txt](https://gist.githubusercontent.com/jfwells/c0aa7a368abc23a3075e45692ab2fb17/raw/cfe5ff91ad8b14b5929773606aeb693f40f10c1e/debug.txt])

Oddly enough, msfs2020 seems to detect that TrackIR is available (at least, the slider is toggleable, which it wasn't before -- but no data makes it through). wine::pose is still firing though, despite the module not being loaded.

De-Clan-C, 2024-10-29:
> * I kept E-SYNC/F-F-SYNC enabled and used the Freetrack protocol.

Changing to the freetrack protocol made it work for me. I think we should get this information into a wiki page so people don't have to come looking for this issue in order to find it. Specifically that your game needs to have it's prefix (or a simlink to its prefix) in the default steam compdata folder and then you set the game ID in opentrack and make sure the protocol is set to freetrack.

I would be glad to make a page, but I just wanted to check if anyone else has started one already.

Also, besides what I've mentioned here, is there anything else that should be mentioned?

Breeffeehey, 2025-04-11:
> I'm a bit new to opentrack. I've taken a different (hacky) approach to quickly getting some head tracking working in IL-2 BoX: I translate opentrack UDP-output to mouse (and mouse-wheel) moves. IL-2 BoX can't distinguish these evdev/HID events from real-mouse moves. There's more discussion in [issue 1560](https://github.com/opentrack/opentrack/issues/1560#issuecomment-1345480710) describing my progress. I uploaded my experiments to a github project [digitaltrails/opentrack-opal](https://github.com/digitaltrails/opentrack-opal).
> 
> I've also created a UDP-output to joystick - but IL-2 BoX doesn't seem to play well with that for reasons I don't understand. The game recognises the stick, but can't seem to map any axes - maybe I need to emulate a "known" stick, maybe I need to edit some IL-2 BoX config somewhere (I had to delete devices.txt to get IL-2 to recognise it at one point).

I'm bowing down, kneeling to you!
I've been digging the whole web to make AITrack and OpenTrack work with ETS2 in linux (I'm on KDE Neon) and all I got was windows setup or videos of people showing it working...
Thanks to you, I got it tracking, now! Next step is to tweek the settings so I don't feel like breaking the neck of my avatar when I look around... (and keybinds to start/stop/pause tracking)
Thanks again!