opentrack/opentrack #1083: Only 30Hz with Webcam and Aruco?

BenGohan, 2020-05-09:
I am using opentrack with Aruco. It works great but the camera window in the program shows 30Hz no matter what actual frame rate the webcam is set to or is specified in the open track settings. I am trying various cams but still I only see 30Hz. I would be very happy to hear whether higher poll rates from the cam could be achieved, since it would increase the overall tracking smoothness within games. Thanks a lot.

sthalik, 2020-05-10:
What webcams are you using?

BenGohan, 2020-05-10:
Thanks for the questions, it made me recheck the cameras: The Creative HD 720 is only 30Hz and I think the same goes for the "DroidCamClient" I use for using mobile phones as webcam under W10. The phones support higher framerates and resolution but I guess the "DroidCamClient" is limiting factor.  I guess I will have to buy a decent (and recent) one. Thanks again.

closed 2020-05-10 by BenGohan