opentrack/opentrack #1086: PS3 Eye Camera FPS Problems

Khangirey, 2020-05-17:
A quick question before I begin - In the preview, is it supposed to preview the max fps? Even if I set it well above 20, it seems to hover around that mark. I don't know if this is a bug or intended. 

1. I have the PS3 Eye Camera with the updated CL drivers. At 640x480, I am only able to get a max of 70hz. If I go above that, there is no video feed and the screen is black. 

2. What should my diagonal FOV be sets to for the Eye cam?

Khangirey, 2020-05-17:
Addendum - If I set the resolution to half in an attempt to get 120 fps. I am unable to set it anywhere past 60.

sthalik, 2020-05-18:
> What should my diagonal FOV be sets to for the Eye cam?

56 or 75.

Khangirey, 2020-05-18:
@sthalik 

Thanks. I just found something else out. Even at the default resolution, when I go past 60, the status readout shows a lower FPS than if I were just to leave it at 60. Ex: Putting it to 70 shows FPS at a max of 55.

MathijsG, 2020-05-20:
I have the same problem with the current version.

sthalik, 2020-05-24:
How's CPU usage?

pyrob2142, 2020-05-27:
Just want to leave a note that I'm experiencing the same issues, though I doubt that they are related to opentrack, as these issues are within CL-Eye Test as well. Also if you look at the recording LED, maybe you can confirm, that it starts flickering instead of staying a solid red. This happens if I increase the frame rate above 60 Hz regardless of resolution.

MathijsG, 2020-05-27:
I don't have this problem with the latest beta, but I do encounter this
problem with the stable version.

Groeten,

Mathijs Groothuis

Khangirey, 2020-06-26:
I should add that randomly I was able to get the 75hz. I reinstalled Windows the other day, and I'm back to the previous problem. If I find exactly how to do so, I will post my findings.

Khangirey, 2020-06-26:
I FOUND IT!

You need to be plugged in directly to the motherboard, and using USB 3.0

There is however another problem(?) that occurs. While it does properly run at 75hz, there are now these weird vertical lines over the image. They aren't lines per say, but more of color discoloration that runs in a line. With an IR filter however, I see no problem at all.

pyrob2142, 2020-06-26:
That's interesting. I'm using it at an USB 3.0 port on my Crosshair VI Hero and am only able to get a picture at a maximum of 60 Hz, same as with a USB 2.0 port, every other refresh rate produces a black screen / no signal.

sthalik, 2020-06-26:
@Khangirey 

> While it does properly run at 75hz, there are now these weird vertical lines over the image.

Lower gain and the distortion is minimal.

Khangirey, 2020-06-28:
@sthalik 
Do you know the exact reason for this?

@pyrob2142 
I had to fiddle with it too in terms of ports. The USB 3.0 Ports via my case also give me a black screen. I think it might depend what USB 3.0 Gen it is.

PredatorNo1, 2023-02-15:
Hello Guys,

I finally found the solution for this. I've been using my TrackIR system for a long time, I always had the stable 75FPS. Recently I had to reinstall the windows from the bottom, and the FPS drop appears. I couldn't set it to 75FPS not above 70 even then I get only 56FPS. After a while, I tried to reinstall all the drivers, install all the updates, literally tried everything.
The mistake I made, I plugged the PSEYE3 into my motherboard USB 3,2 instead of 3.0.... I redo it and it's working immediately.
Have a good day!