opentrack/opentrack #1087: OpenTrack - ?Disabling rather than uninstalling to release joystick and keyboard?

JBlyth, 2020-05-18:
I am having problems with OpenTrack seeming to "grab" joystick controls and some keyboard mappings, and the only way I can "release" them is to uninstall OpenTrack to regain normal joystick and keyboard control for my game if I wish to "fly" without OpenTrack.

I am playing Condor 2 Gliding Simulator. How can I get OpenTrack to release the keys without having to uninstall it. I cannot find any running processes to OpenTrack, but it must be locking things somehow.

Cheers
John in Sunny Carnarvon - Western Australia

Len62, 2020-05-18:
Look in the options, shortcut tab, and see what, if any, bindings are there.

JBlyth, 2020-05-18:
Thanks Len2,

I did have two bindings there for the trigger and one other button, but even when I released those, the number pad assignments and all others still seemed to be locked out.

When I uninstalled OpenTrack, they were all released again.

Cheers
John

Len62, 2020-05-18:
So just having OT installed locks out those items, even with OT not running? What are you using for input and output? You could uninstall and then try the version w/o the installer. Just unzip it to a folder of your choice.

JBlyth, 2020-05-18:
That is correct. Without having OT running in anyway, all keyboard and joystick keys relating to screen movement that were replaced by OT were disabled with Condor 2. When I removed OT, the keyboard and joystick function all immediately returned to their in-game settings.

I will try again tomorrow with the unzip to a folder method and let you know how I go. Fingers crossed!

Cheers
John in Sunny Carnarvon - Western Australia

sthalik, 2020-05-18:
It's not the keys. It's game functions related to moving the camera. Delete or rename NPClient.dll and NPClient64.dll to fix this.

JBlyth, 2020-05-18:
Thanks, sthalik,

I just tried it from the zipped version, and had exactly the same result - many keys "locked" by OT.

Then I deleted that OT folder and re-installed in the normal way. I deleted the NPClient.dll and NPClient64.dll files that you mentioned. The usual joystick and keyboard responses were still present, but sadly, although OT showed the gyroscopic input from my phone, and in the game side of the data too, there was no response from Condor 2 at all from OT.

Note that I am using the Android app HeadTracker as a source of gyroscopic data as the one supplied in OT doesn't load on my Motorola phone. My laptop is Windows 10 and all updates are installed.

Happy to help in any way....

Cheers
John

sthalik, 2020-05-18:
> although OT showed the gyroscopic input from my phone, and in the game side of the data too, there was no response from Condor 2

If you're changing the installation directory, you need to start tracking once, before starting the game. Now it's all fine since first start updated the registry entries.

JBlyth, 2020-05-19:
I have done some extensive testing, and here is what I have found:-

- Renaming NPClient64.dll has no effect at all, even though I am running a 64 bit Windows 10 laptop.
- Having NPClient.dll available gives me OT control, but the game's keyboard and joystick functions related to video are disabled.
- If I rename NPClient.dll there is no OT connection to the game at all, although OT still shows data incoming and on the Game section as well, but I get back keyboard and joystick functions - but only after a reboot.

I have rebooted between each scenario.

I hope this helps track down the issue.

Cheers
John in Sunny Carnarvon - Western Australia

jfitter, 2020-06-09:
Same problem with Condor 2 using edtracker and win7. Software MUST be able to unload itself. It appears OT is incapable of this. I have uninstalled it and it will remain this way until OT moves from toy status to pro status. I suggest removing Condor from the supported games list - it is NOT supported.

alistairmgreen, 2020-07-26:
I had the issue with Condor as well, and fixed it by using Regedit to delete the following registry key which OpenTrack left behind:

```
HKEY_CURRENT_USER\Software\NaturalPoint
```
It would be nice to have some way of doing this through the OpenTrack GUI.

As an aside, I am using the "portable" version of OpenTrack and therefore did not expect it to make permanent changes to the system like this!

See also #1115

JBlyth, 2020-07-27:
Thanks Alistair. I tried that without success. Perhaps is is because I am using the UDP Source. 

My only solution so far is to use FaceTrackNoIR.

Cheers
John in Sunny Carnarvon - Western Australia

vggoggi, 2020-11-08:
John, I have the **very same** problem and experience with Condor 2. No help received so far. This issue seems to have reached a dead end. 
Given your experience since end-July (your last post), can you please confirm that using FaceTracknoIR the behaviour of OT gets back to "normal"? Any further advice?

Many thanks and cheers,
Giorgio in Pavia Italy

JBlyth, 2020-11-08:
Hi Giorgio,

I haven't used OpenTrack for a few months now as I found too many issues
when using it with Condor 2.

FaceTrackNoIr mightn't look as polished, but I find it works really well
for my needs.

The developer is also very responsive and helpful with any issues.

Cheers
John in Sunny Carnarvon, Western Australia

JBlyth, 2020-11-08:
PS. I can confirm that stopping FaceTrackNoIr does immediately release the normal keyboard/joystick functions.

I am now using a Thrustmaster Rudder/Throttle/Joystick pack these days (about$400 Australian) and it is truely brilliant when setup with Joy2Key and FaceTrackNoIr!

I'm happy to share my setup files if you are interested. I also use a Declan Fusion headtracker which is also brilliant!

vggoggi, 2020-11-08:
Hi John,
thanks for your very prompt and clear reply!
In fact, I had dropped using OT for months, given the known bug plus the difficulty in controlling motion smoothly (my settings, not an OT bug). I resumed yesterday and found new better settings (EWMA filter set to very slow, and so on).

Given your report, I plan to switch to FaceTrackNoIR soon and would VERY MUCH APPRECIATE your settings, given my slow/poor learning curve so far. Screenshots (or other, as you prefer) would be fine.

At the moment I have achieved 2 "steady" position with smooth transitions (in Yaw): forward (dead zone) and at +/-90° (end of curve). I have been trying to add semi-steady views at about +/-45° with a step in the curve, but the head range movement becomes sub-optimal and steep segments become necessary (implying some fast rotations, unpleasant to the eye). Sure your settings will be better. 

Have been a DG 200 owner and glider pilot in real life for a lifetime. Now 78y, had to give up licence and Condor 2 came to rescue ...
Thank you so much,
Giorgio
"Think positive, flaps negative !"

JBlyth, 2020-11-09:
Hi Giorgio,

Have a look at my post with two attached setting files at:-
https://sourceforge.net/p/facetracknoir/discussion/1150909/thread/af466014cb/

I used to have a share in an early model ASW22 in Australia for a decade or more. Those beautiful 24 metre wings were a delight! Last year I was very fortunate to have the opportunity to fly at Kiripotib in Namibia for 8 weeks in an Arcus M. For the past 22 years I have also towed gliders in northern Japan for 2 months each year, but sadly Covid stopped that this year......

Cheers
John

vggoggi, 2020-11-09:
Hi John,
thanks again for your reply and help.
Wow, what a great gliding experience you have! I fly occasionally Open class on Condor 2 only, beautiful gliders but somewhat inert in roll.
COTASA is also very useful, used on some cross-country flights in the Andes (up to 1000km).
In the end, however, I prefer the DG 808-18, more reminiscent of my beloved '200.

Sorry to bother, but what should I do with the CondorDelan1.ini file? (Please note that I have not approached FaceTrackNoIR yet).
All the best,
Giorgio

JBlyth, 2020-11-09:
Hi Giorgio,

It is my .ini file for FaceTrackNoIr. I mis-named it a little. It Should
really be Condor Declan1,ini to be clearer ;-)

If you open it from within  FaceTrackNoIr, it will give you all the
settings that I use.

Cheers
John

vggoggi, 2020-11-09:
Perfect, John, thanks!
I really appreciate your help. Will begin exploring FaceTrackNoIR.
Happy flights,
Giorgio