opentrack/opentrack #1088: Getting rid of the limit for coordinates

JabberwockPL, 2020-05-21:
I am trying to do a relatively simple thing: I want to receive UDP data from FreePIE on another computer and output them (with no changes at all) as Freetrack (for head position in VRidge). However, I have hit a snag: the output is arbitrarily limited to 100 cm, even if the limit in the .ini is set to a much higher value.

Could it be made so that the limit works only if there is no overriding value in the .ini?

sthalik, 2020-05-24:
> the output is arbitrarily limited to 100 cm

The output methods (i.e. flight simulators) don't expect arbitrarily large values. You surely could stuff a large value in there, only for the sim to react unexpectedly.

JabberwockPL, 2020-05-24:
If the excess values were possible only after manual edition of the config file, I think it could be safely assumed that the user knows what he is doing...

However, it turned out that the issue is not relevant after all - the values are actually multiplied by ten when output to freetrack (i.e. what is sent as 30 over UDP is 300 in freetrack), so the values are sufficient, although somewhat imprecise. As the UDP values do not allow fractions, effectively I can only send the position in 10 cm increments. I can live with that, I guess, so I am closing the ticket.

closed 2020-05-24 by JabberwockPL

sthalik, 2020-05-25:
> If the excess values were possible only after manual edition of the config file, I think it could be safely assumed that the user knows what he is doing...

The values are stored precomputed in the mapping editor. Mapped values outside the range don't exist at all.

> As the UDP values do not allow fractions

That's not true.

JabberwockPL, 2020-05-25:
Yes, you are right, I was misled by the displayed values. UDP values of 30.4 and 30.6 in the UI are represented as raw/game values of 30 and 31, but actual freetrack values are correct: 304 and 306. Great!

sthalik, 2020-05-25:
> actual freetrack values are correct: 304 and 306.

These values are all floating-point. You have at least 6 decimal points of precision.

lh-dm, 2022-02-22:
This is very important for me.

I would like to use opentrack for a exhibition.
The room is much bigger than normal and also the monitor is much bigger than normal. I need to be able to track large distances. 
For example Z depth can be 5m and X direction can also be 5m.
Camera should be placed at a height of 2.8m!

I see the software is only limited to 100cm - and that seems to make it not usable. I am a bit unhappy about this.

The only workaround I can see would be to artificially mis-size the cap dimension (not tested).

If freetrack is wrong solution (i need only XYZ-position), what software can i use instead?

sthalik, 2022-02-22:
It won't do you much good without making an enormous cap. The points are too small beyond a meter or two.