opentrack/opentrack #1091: open track not seeing led dots?

Vegas666, 2020-05-28:
Hi!

I am very new to this. I started playing warthunder with open track and ps3 camera last year. I went more  than 8 months without playing the game or using opentrack. Today I decided to come back but opentrack doesnt work...not just doesnt work in game, but doesnt work at all.

it seems opentrack is not connecting to my camera, as the interface shows just a black screen without the tracking dots. When I go to pointracker settings it shows device as ps3 camera, 640*480, fps 70hz, Status (at the bottom) says Camera info: 640*480 @ 0pfs Extracted points: 0 Bad! 

The camera still works, and with the ir screen (CL-eye Test) still shows the tracking leds, therefore camera and headset are working,I hadn't changed anything since I last used it when it was working properly.

Since the problem started, I uninstalled opentrack 2.3.10 and installed Opentrack 2.3.11 software,  but still with no luck.

I read somewhere else that AVG anti virus was the cause for some people but I don't have it on my computer...

Really apreciate the help =)
[image]

JacekJagosz, 2020-05-30:
Could you tell if the red LED on the left turns on when you start tracking?
Also, make sure no other app using a camera is running in the background, if you have started CL Eye Test before the tracking it could've caused the problem. 
Other than that I don't have any good idea apart from trying to reinstall CL-Eye driver

Vegas666, 2020-06-01:
Hi!

Thank you for answering. I realized the other day it was because I had CL Eye open... it had been so long since I last used this I totally forgot about closing the app before using open track.. I feel so stupid =P

closed 2020-06-01 by Vegas666