opentrack/opentrack #1092: Opentrack GUI

K3NART, 2020-05-31:
Whilst setting up opentrack I got as far as setting up the curves when the user interface disappeared from  my screen. Closed the software in task manager and re-opened but no user interface appears although task manager states it is running. Uninstalled and re-installed but still the same problem.
I cannot get the user interface to appear on screen so I am unable to make any changes.
Windows 10 OS. 

Opentrack running but no interface on screen
[image]

sthalik, 2020-06-01:
Do you have tray support enabled?

K3NART, 2020-06-01:
Do you mean system tray.
If not can you give me more details/instructions please
I'm a 62 yr old male and know very little about windows 10 :(

K3NART, 2020-06-01:
All sorted thankyou.
It was in the system tray.

sthalik, 2020-06-02:
Glad it works for you now.

closed 2020-06-02 by sthalik