opentrack/opentrack #1097: FreePIE UDP axes/planes affecting each other

dglava, 2020-06-09:
I want to use my Android smartphone with the FreePIE UDP tracker for headtracking. I set it up and it's receiving data.

The problem is that movement on the pitch axis seems to somehow trigger the other 2 and the Octopus moves in an unwanted way. Example: I look down and it translates it into looking down and to the side. When I only assign pitch and disable yaw and roll, looking down and up works fine.

Video of the undesired movement: https://s1.webmshare.com/6V6Bj.webm
Another video of the undesired movement: http://s1.webmshare.com/WxONP.webm

Another interesting thing is: when I position the phone on a table and turn it until the raw tracker data for yaw says approximately 0 and then start tracking, it seems to work, i.e. looking down doesn't result in looking down/sideways/rolled. It's kind of annoying though to always position the phone in a specific way and then pick it up again.

[Screenshot](https://i.ibb.co/JjKmVcH/2.png) of the tracker data when it works, i.e. when moving head forward doesn't make it yaw or roll.
[Screenshot](https://i.ibb.co/hyRknTn/1.png) of the raw tracker data when I align the phone how it's on my head and when tracking results in unwanted yaw/roll when pitching.

I looked through the previous issues and  this one #227 seems similar. I tried the suggestions (adding to the axes), but it didn't help.

Is this something to be expected from Smartphone tracking? Can it be fixed? I reckon that if there was a way to "restart" or center the raw data, i.e. tell the phone "this is the starting position", that it would fix this.

Phones: Samsung Galaxy S7 and Samsung A20e 
Opentrack version: 2.3.11

sthalik, 2020-06-17:
I can't see the raw yaw values during pitch. But they seem to be varying a lot.

dglava, 2020-06-22:
Is it due to the nature of the phone's sensors or do you think it's a problem in Opentrack? I guess I can close this issue if it's something that can't be really fixed on Opentrack's side.

But if you think it can be improved, I can post additional information, just tell me what you need.

sthalik, 2020-06-23:
> Is it due to the nature of the phone's sensors or do you think it's a problem in Opentrack?

It could be.

> But if you think it can be improved, I can post additional information, just tell me what you need.

What is the difference between yaw values between phone lying flat on the surface, and rotated up? Please do this experiment when centering, and also when "center on startup" is disabled and centering not done at all.

dglava, 2020-07-17:
Since I have switched to point tracking now, I haven't been troubleshooting this much. I will close this for now, but perhaps see if I can provide more data in the future.

closed 2020-07-17 by dglava