opentrack/opentrack #1099: PS3 Eye FPS Issue

sj729, 2020-06-09:
Hello,

I am trying this DIY TrackIR thing out and I have everything set up, or so I think. 
My PS3 Eye is getting weird framerates, it jumps from whatever I have it set to to any where from 0-5 then to 10-20 then repeats the process. This all happens within a second and every second. Im not sure whats up, I have Pointtracker 1.1 and freetrack 2.0 enhanced enabled with default settings. The tracking is all over the place, it stutters so much that it's unusable. Any thoughts?

MicrowaveBandit, 2020-06-09:
Im having this issue as well. The FPS counter for the PS3 cam constantly changing between 30 and 0. This doesnt happen when im testing the cam outside Opentrack and i have confirmed that the IR lights on the headset arent flickering.

sj729, 2020-06-09:
After doing some more research, someone else had the same problem on this repository and reported that the problem was fixed after a few windows updates. Low and behold, after a quick update, the stuttering was gone and the fps was solid at whatever I set it to. Very weird, but it works great now! @MicrowaveBandit, I would try this first if you haven't already... if not I will leave the issue open.

MicrowaveBandit, 2020-06-09:
Hi,
Just updated windows and the problem appears to be solved.
Thank you so much for your help @sj729

sj729, 2020-06-10:
@MicrowaveBandit glad I could help!

closed 2020-06-10 by sj729