opentrack/opentrack #1100: Crashing when clicking start using PointTracker 1.1 input

jsandr, 2020-06-09:
**Issue**
When I click "Start",  OT closes (sometimes after a few seconds freeze). I do not have any video feedback in the meantime.

**OpenTrack Settings**
-  Running as Administrator?: Yes/No
- Input: PointTracker 1.1 (Set to use PS3 Eye Camera)
- Output: freetrack 2.0 Enhanced
- Filter: Accela (All settings default)

**Software**
- OT 2.3.11 (installed at C:Programmes(x86)/opentrack)
- Win 10 v.1906
- CL-Eye-Driver-5.3.0.0341-Emuline

**Hardware**
- Modified PS3 Eye
- DelanClip

**Details**

- I've tried with differents input options, with wich OT doesn't crash anymore. However I can't tell if OT is working properly afterwards because I'm not set up to use thoses.
- The CL-Eye Test is working, I can see the 3 leds alright.

sthalik, 2020-06-09:
That's a PS3 Eye driver bug. Try plugging the camera into another USB port.

jsandr, 2020-06-10:
**Edit**
When using the integrated Webcam of my computer, OT is still crashing.

- I've launched FreeTrack, and I can track the 3 points on my clip using that intergrated webcam, I have video feedback. In short : it works.

So going back to OT :  no matter wich input hardware I'm using, OT cannot handle it.

- Driverswise, my integrated camera is up to date and the PS3 Eye are uninsralled and my pc rebooted.

Also when going clicking Open on "Camera settings" OT crashes. Settings that are available using FreeTrack.

jsandr, 2020-06-10:
> 
> 
> That's a PS3 Eye driver bug. Try plugging the camera into another USB port.

Nice try, but thats going nowhere.

**Update**
When using the integrated Webcam of my computer, OT is still crashing.

- I've launched FreeTrack, and I can track the 3 points on my clip using that intergrated webcam, I have video feedback. In short : it works.

So going back to OT : no matter wich input hardware I'm using, OT cannot handle it.

- Driverswise, my integrated camera is up to date and the PS3 Eye are uninsralled and my pc rebooted.

Also when going clicking Open on "Camera settings" OT crashes. Settings that are available using FreeTrack.

sthalik, 2020-06-10:
Send me the minidump. While the software is crashed, right-click the process in Task Manager and select "create dump file".

norsetto, 2020-06-20:
In my case it always crashes if I first stop it and the start it again. zipped dump is unfortunately quite big: can try with a debug build once I download the source tough.

norsetto, 2020-06-21:
Compiling from current head I get no crashes anymore. As a nice bonus, also tracking seems to be much better 🥇