opentrack/opentrack #1101: Yaw inverted with Joystick input full left

rsys-dev, 2020-06-10:
Hi,

I installed Opentrack yesterday to use the Analog Stick of my Fanatec Rim to look around in the car.
Generally it seems to work fine, but if I go full left with the stick, the Yaw Value gets negated which results in suddenly looking fully right instead of left in the game.
This happens when RAW Input Value hits 180. Maybe the issue is also on fully right, but on the right side the maximum RAW Input Value is 177, so I can't check if it is an issue only on the left side or also on the right side.

Attached you can find my current mapping, but it also happens with default mapping. Also changing Max Input changes nothing as it seems not to limit the real input but only shapes the graph.
[image]

Anything I missed in the config or any idea how I can fix this?

Kind regards
Manuel Ritter

sthalik, 2020-06-12:
What happens if you disable auto-center? Is it acceptable for you to not use opentrack's center functionality with the analog stick?

rsys-dev, 2020-06-13:
After I disabled "Center at Startup" the issue is gone.
Thats fine for me, thank you.

closed 2020-06-13 by sthalik