opentrack/opentrack #1102: Opentrack has different title for game and doesn't respond correctly.

BuTTuRz, 2020-06-11:
Hi guys.  I'm new to this site and I'm only here because I"m not sure where else to go with the issues I'm having.  So for most games, Open track works flawlessly.  For these two specific games though for me, they don't work correctly.  Dirt Rally 2.0 and Assetto Corsa.  When I launch Open Track and try to play these games, it comes up as the wrong game title for both.  The tracker seems to lock up and doesn't recognize the inputs.  They seem to jitter and bounce back and forth.  When I close the game, boom. Open Track works normally again.  I always run it on Admin.  It's allowed in firewall.  Works great for Elite Dangerous, Star Citizen and Project Cars 2.  Just doesn't work at all for those 2 games listed above.  Hoping someone here can help me fix this.  Thanks in advance ya'll.

BuTTuRz, 2020-06-12:
Bump.  I'm still having issues and I still can't seem to fix them.

sthalik, 2020-06-12:
> The tracker seems to lock up and doesn't recognize the inputs. 

Can you elaborate on that? It also reminds me of another issue regarding Game Mode in Windows 10.

BuTTuRz, 2020-06-12:
https://youtu.be/6RJc9n6Tjpo   I've just recorded the issue as it happens.  Also, take notice that the title of the game is different than the actual game I'm playing.  (I'm playing Dirt RALLY 2.0.  Not DIRT 2)

BuTTuRz, 2020-06-12:
And you saying game mode made me take a look at mine, and I had it enabled.  Disabled it and the issue still persists.  Figured I'd say that upfront.

sthalik, 2020-06-12:
> wrong game title

Lots of games reuse the same identifier. That's not the cause of this issue.

> The tracker seems to lock up and doesn't recognize the inputs.

I think this is caused by a feature that's meant to save CPU time, but is broken on multiple displays. Namely, opentrack tries not to refresh its main UI (but refreshes the video feed, which I only recently fixed). This only works reliably with a single display which is why it behaves like that. I just fixed it in d64e1ec0a5.

If you want to reproduce this, position both the game and the opentrack window on the first screen and see the numbers change.

> Assetto Corsa

I tried googling it and someone mentioned making sure to use cockpit camera. There are apparently lots of people using EDTracker with the game.

BuTTuRz, 2020-06-12:
So, is there anything I can personally do with this information?  If I were to disable my 2nd monitor while starting it up, would that possibly fix the issue?  Also, as it stand, in the newest released version, If I have open track on my first monitor (the same where the game is displayed) the issue still happens.  Not sure if that information helps at all.

BuTTuRz, 2020-06-12:
Follow up.  I've unhooked one of my monitors to fix a single monitor setup and it still froze as before.

BuTTuRz, 2020-06-13:
So, after some more hours of messing around.  I've narrowed the problem down to the Wireless USB dongle of my Xbox One Controller.  When I take it out and hard wire the controller, the headtracking works as it's supposed to.

sthalik, 2020-06-17:
There's nothing opentrack can do about one USB device interfering with another. Thanks for reporting back the solution.

closed 2020-06-17 by sthalik