opentrack/opentrack #1104: vjoystick not working in 2.3.11

ODMGamer, 2020-06-17:
Good afternoon! I've used OpenTrack with great success on Linux many times, and thought I'd have similar success using it on Windows, but have immediately bumped into some trouble. When I select joystick emulation after having installed the latest version of vJoy, I get this error:

[image]

I've tried installing the previous version of vJoy with the same results, a driver/SDK mismatch.

My opentrack version:
[image]

My vJoy version:
[image]

I'm presently on Windows 10 Home 64-bit version 10.0.19041 build 19041. What additional information do you need from me to help troubleshoot this issue? Thanks!

ZiCVIZ, 2020-06-17:
I confirm ... I have the same thing

kilrah, 2020-08-19:
Same. What vJoy version should we be using?

gobusnuts, 2020-08-25:
I can confirm same issue with vJoy v2.1.9 and opentrack 2.3.12 on win 10  64 bit build19041