opentrack/opentrack #1105: Bug with CENTER shortcut and startup minimized!

Shadoware, 2020-06-17:
Im using OpenTrack 2.3.11 With Windows 10 64bit 1909, and this software is great but Im having some bugs:

1 - Sometimes it starts minimized even with this option disabled;

2 - When I use the center shortcut, either in game or not, the raw tracker data continues but game data stills in zero, so the game view becomes immovable. Im using "restart tracking" as a workaround.

Shadoware, 2020-08-08:
I found that if I use "restart tracking" many times the OpenTrack crashes!

sthalik, 2020-08-09:
This is a PS3 Eye driver bug.

sthalik, 2020-08-09:
> Sometimes it starts minimized even with this option disabled; The tick box seems inverted.

This was fixed in unstable.

> When I use the center shortcut, either in game or not, the raw tracker data continues but game data stills in zero

This was fixed as well. As a workaround, don't use a modifier (e..g shift+C), only the key.

closed 2020-08-09 by sthalik

Shadoware, 2020-08-11:
The unstable version is already compiled? Can I just download it and use it?
Im using the version without installation.

sthalik, 2020-08-20:
I've released 2.3.12.

Shadoware, 2020-08-21:
Awesome man!

Shadoware, 2020-08-21:
The related problem about crash dont seem to happen if I dont use restart tracking shortcut even with ps3 eye but perhaps I try this new open driver support another time.
Many thanks, your software helps a lot all the flight sim world!