opentrack/opentrack #1106: wrong path in modules load

grillo78, 2020-06-18:
you have some paths wrong, I'm getting this when opentrack try to load the modules

`DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "rift-025" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-025.dll: The specified procedure could not be found."
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "rift-042" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-042.dll: The specified procedure could not be found."
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "rift-080" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-rift-080.dll: The specified procedure could not be found."
WARN DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("DejaVu Sans", lfWidth=0, lfHeight=-11) dpi=96`

sthalik, 2020-06-19:
The path in square brackets? It's my local build path. These Rift modules are obsolete and should have deleted when upgrading. Looks like there's something wrong with the installer.

grillo78, 2020-06-21:
no, I mean the module path, it has \\\\ instead of \\

sthalik, 2020-06-24:
All `////` or `\\\\\\` are treated the same as `\`.

grillo78, 2020-06-24:
so why can't opentrack find the modules? the DLLs are in the folder, I really think that the program is trying to get it from modules\\module_file.dll

sthalik, 2020-06-25:
Must be a missing library. What Windows version is it?

grillo78, 2020-06-29:
10.0.18363

sthalik, 2020-07-01:
Actually these Rift libraries are obsolete. Try doing a fresh install after deleting the install directory. For Kinect it's normal for the load to fail when not having the SDK installed.

grillo78, 2020-07-02:
I have the SDK installed

sthalik, 2020-07-02:
Is Kinect the module you're interested in? Then show me the output of Dependency Walker on the Kinect module.

grillo78, 2020-07-02:
what do you mean? I can't access to the kinect plugin because opentrack CAN'T find it

sthalik, 2020-07-02:
Of course it can.

The specified **procedure** could not be found.

grillo78, 2020-07-03:
so, how can I get access to the kinect module?

sthalik, 2020-07-03:
Install the Kinect SDK. If it doesn't work, use Dependency Walker.

WangGakin, 2026-05-01:
hello that's a old issue(2020), but i found it today haha (in win11 home edition),  and i followed sthalik - use Dependency Walker run it, and found the reason: my system losts some file from Visual C++ Redistributable packages, so i installed both x32 and 64 packages then everything is fine now. Just for remind.