opentrack/opentrack #1111: Buggy yaw behaviour for joystick input

vin9x, 2020-06-27:
I have mapped yaw to one axis of a gamepad thumbstick and for some reason it randomly flips signs around the outer input values.

xyz4tech, 2021-07-20:
Having the same issue. Any additional information needed for analyzing the issue?

aphor, 2026-10-31:
I just built a custom device that provides a BLE HID 6 axis joystick. All the rotation axes are already filtered and smoothed, verified in control panel. Pitch and roll axes seem to work fine. Yaw input data looks great in control panel, but looks like it's clamped to some tiny range in OpenTrack. It moves, but doesn't provide useful values, certainly not what Control Panel game controller inspection shows.