opentrack/opentrack #1113: Linux, no outputs, "Library load failure".

LevitatingBusinessMan, 2020-06-30:
Hello,

I've been trying to get opentrack to run on my linux machine.

I am using the opentrack-git package.
These are the install params.
```
  cmake .. \
      -GNinja \
      -DCMAKE_BUILD_TYPE=Release \
      -DSDK_XPLANE=xplane_sdk \
      -DSDK_WINE_PREFIX=/ \
      -DCMAKE_INSTALL_PREFIX=/usr
  ninja
```

The full script is here:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=opentrack-git

Currently everything builds but opentrack only has PointTracker as input and Accela as filter. No outputs.

I understood that I needed to set `SDK_WINE` to be able to get the wine-plugin working, and so I messed a bit with ccmake. But ideally I want the aur package to work, so the variable should be set immediately. I tried adding `-DSDK_WINE=ON \` but it didn't work. And altering the var with cmake didn't work either.
And I am not sure if `/` is the correct value for SDK_WINE_PREFIX,
but that variable also doesn't show up when I build manually and run ccmake.

Quick extra question, will opentrack ever see direct support for Face_API like FaceTrackNoIR or something similiar?

sthalik, 2020-06-30:
The `SDK_WINE_PREFIX` variable is obsolete and has been removed. You need to set `SDK_WINE` and have `winegcc` in `PATH`.

Show me `ldd` output for few of the libraries.

LevitatingBusinessMan, 2020-06-30:
[ldd](https://termbin.com/vh3c)

`winegcc` is in the path.

I build manually, I used ccmake to toggle the SDK_WINE value.

```
cd projects/opentrack
mkdir build
cd build
cmake ..
ccmake .
make
make install
```

In `ccmake` I only toggle the SDK_WINE, then press c and g.

sthalik, 2020-06-30:
So `make opentrack-proto-wine` runs but it doesn't work at runtime? And you're missing other output methods including UDP? Use `LD_DEBUG=libs,symbols` and show me the relevant bits. And what is the library load errno (included in the checkbox and stderr).

> Quick extra question, will opentrack ever see direct support for Face_API like FaceTrackNoIR or something similiar?

I should eventually support clandmark for face tracking, but stability and quality is always worse than the reliable methods. But see #1095 for an alternative.

LevitatingBusinessMan, 2020-06-30:
@sthalik I did not realize I had to run `make opentrack-proto-wine` seperately.
Am I missing other build steps?

My opentrack looks like this: https://i.imgur.com/4IZvNA8.png
No outputs.

sthalik, 2020-06-30:
> And I think I misunderstood, just make should build the wine protocol right?

Yes, but opentrack won't work unless you run the `install` target.

> The error found in stderr: DEBUG [../logic/runtime-libraries.cpp:25]: protocol dylib load failure

What's the contents of `prefix/libexec/opentrack`? Aren't you getting load failures on `stderr` for each module separately? The error you posted is when starting tracking, not the initial load that populates the combobox.

LevitatingBusinessMan, 2020-06-30:
```
-rw-r--r-- 1 rein rein  16K Jun 30 00:07 freetrackclient64.dll
-rw-r--r-- 1 rein rein  72K Jun 30 00:07 freetrackclient.dll
-rw-r--r-- 1 rein rein  20K Jun 30 00:07 NPClient64.dll
-rw-r--r-- 1 rein rein  16K Jun 30 00:07 NPClient.dll
-rwxr-xr-x 1 rein rein 722K Jun 30 18:23 opentrack-api.so*
-rwxr-xr-x 1 rein rein 1.6M Jun 30 18:23 opentrack-compat.so*
-rwxr-xr-x 1 rein rein 1.5M Jun 30 18:24 opentrack-filter-accela.so*
-rwxr-xr-x 1 rein rein 1.4M Jun 30 18:24 opentrack-filter-ewma.so*
-rwxr-xr-x 1 rein rein 1.7M Jun 30 18:24 opentrack-filter-hamilton.so*
-rwxr-xr-x 1 rein rein 6.7M Jun 30 18:25 opentrack-logic.so*
-rwxr-xr-x 1 rein rein 5.1M Jun 30 18:25 opentrack-migration.so*
-rwxr-xr-x 1 rein rein 6.1M Jun 30 18:23 opentrack-options.so*
-rwxr-xr-x 1 rein rein 768K Jun 30 18:24 opentrack-pose-widget.so*
-rwxr-xr-x 1 rein rein 1.4M Jun 30 18:24 opentrack-proto-fgfs.so*
-rwxr-xr-x 1 rein rein 1.1M Jun 30 18:24 opentrack-proto-libevdev.so*
-rwxr-xr-x 1 rein rein 1.4M Jun 30 18:24 opentrack-proto-udp.so*
-rwxr-xr-x 1 rein rein 2.2M Jun 30 18:24 opentrack-proto-wine.so*
-rwxr-xr-x 1 rein rein 1.4M Jun 30 18:23 opentrack-qxt-mini.so*
-rwxr-xr-x 1 rein rein 3.0M Jun 30 18:24 opentrack-spline.so*
-rwxr-xr-x 1 rein rein 5.0M Jun 30 18:25 opentrack-tracker-easy.so*
-rwxr-xr-x 1 rein rein 1.5M Jun 30 18:24 opentrack-tracker-freepie-udp.so*
-rwxr-xr-x 1 rein rein 2.1M Jun 30 18:24 opentrack-tracker-fusion.so*
-rwxr-xr-x 1 rein rein 2.2M Jun 30 18:24 opentrack-tracker-linux-joystick.so*
-rwxr-xr-x 1 rein rein 6.6M Jun 30 18:25 opentrack-tracker-pt.so*
-rwxr-xr-x 1 rein rein 1.7M Jun 30 18:24 opentrack-tracker-s2bot.so*
-rwxr-xr-x 1 rein rein 1.1M Jun 30 18:24 opentrack-tracker-test.so*
-rwxr-xr-x 1 rein rein 1.4M Jun 30 18:24 opentrack-tracker-udp.so*
-rwxr-xr-x 1 rein rein 5.9M Jun 30 18:25 opentrack-user-interface.so*
-rwxr-xr-x 1 rein rein 888K Jun 30 18:24 opentrack-video-opencv.so*
-rwxr-xr-x 1 rein rein 1.2M Jun 30 18:23 opentrack-video.so*
-rw-r--r-- 1 rein rein 127K Jun 30 18:23 opentrack-wrapper-wine.exe.so
-rw-r--r-- 1 rein rein 377K Jun 30 00:07 TrackIR.exe
```

sthalik, 2020-06-30:
On line 239 of `api/plugin-support.hpp`, right below

```c++
        for (const filter_& filter : filters)
        {
            for (const QString& filename : dir.entryList({ filter.glob }, QDir::Files, QDir::Name))
            {
```

add:

```c++
qDebug() << "try load" << filename;
```

LevitatingBusinessMan, 2020-06-30:
@sthalik Sorry for the late response,
here is a screenshot of the program and its output:
https://i.imgur.com/K7d9S0e.png

sthalik, 2020-06-30:
OK. So in `opentrack/main-window.cpp:80`, change

```c++
    for (dylib_ptr& x : modules.protocols())
        ui.iconcomboProtocol->addItem(x->icon, x->name, x->module_name);
```

to:

```c++
    for (dylib_ptr& x : modules.protocols())
    {
        qDebug() << "adding proto" << x->name;
        ui.iconcomboProtocol->addItem(x->icon, x->name, x->module_name);
    }
```

LevitatingBusinessMan, 2020-06-30:
```
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-accela.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-ewma.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-hamilton.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-easy.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-freepie-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-fusion.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-linux-joystick.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-pt.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-s2bot.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-test.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-fgfs.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-libevdev.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-wine.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-video-opencv.so"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "Easy Tracker 1.0"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "FreePIE UDP receiver"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "Fusion"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "Linux Joystick input"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "PointTracker 1.1"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "S2Bot receiver"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "Test tracker"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:78]: adding proto "UDP over network"
WARN QXcbConnection: XCB error: 5 (BadAtom), sequence: 468, resource id: 0, major code: 20 (GetProperty), minor code: 0
```

sthalik, 2020-07-01:
You appear to have modified the `modules.trackers()` block rather than `modules.protocols()`.

sthalik, 2020-07-01:
The issue is that there are no further error paths after `api/plugin-support.hpp` that could prevent an output method from appearing. The "adding proto" debug line is directly adding the output methods to the combobox.

LevitatingBusinessMan, 2020-07-02:
> You appear to have modified the `modules.trackers()` block rather than `modules.protocols()`.

With it in the right location:
```
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-accela.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-ewma.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-hamilton.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-easy.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-freepie-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-fusion.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-linux-joystick.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-pt.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-s2bot.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-test.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-fgfs.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-libevdev.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-wine.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-video-opencv.so"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "FlightGear"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "libevdev joystick receiver"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "UDP over network"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "Wine -- Windows layer for Unix"
WARN QXcbConnection: XCB error: 5 (BadAtom), sequence: 468, resource id: 0, major code: 20 (GetProperty), minor code: 0
```

LevitatingBusinessMan, 2020-07-02:
I removed the opentrack package I installed, then rebuild and reran. I now have 0 trackers, 0 proto's and just Accela as a filter.

```
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-accela.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-ewma.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-filter-hamilton.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-freepie-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-fusion.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-linux-joystick.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-s2bot.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-test.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-tracker-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-fgfs.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-libevdev.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-udp.so"
DEBUG [/home/rein/projects/opentrack/api/plugin-support.hpp:240]: try load "opentrack-proto-wine.so"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "FlightGear"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "libevdev joystick receiver"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "UDP over network"
DEBUG [/home/rein/projects/opentrack/opentrack/main-window.cpp:82]: adding proto "Wine -- Windows layer for Unix"
WARN QXcbConnection: XCB error: 5 (BadAtom), sequence: 468, resource id: 0, major code: 20 (GetProperty), minor code: 0
```

sthalik, 2020-07-02:
Basing on the first screenshot, is this a tiling window manager? This really shouldn't affect anything, but can you try with a normal WM?

As you can see from the last debug log, the modules are actually added to the combobox. There are there. I don't know what to say.

LevitatingBusinessMan, 2020-07-02:
@sthalik I am using i3wm, I was already planning on trying to build it in a clean system. I'll try building on Pop OS today or tomorrow.

LevitatingBusinessMan, 2020-07-03:
```
/usr/bin/ld: relocatable linking with relocations from format elf64-x86-64 (/usr/lib/x86_64-linux-gnu/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.Nx3ZVF.o) is not supported
```
I get this error when building on a fresh PopOS! install.
I have the wine32 and wine64 packages installed.
Though installing wine32-tools causes some weird dependency errors.

```
The following packages have unmet dependencies:
 wine32-tools:i386 : Depends: libwine-dev:i386 (= 5.0-3ubuntu1) but it is not going to be installed
```

I honestly don't know what's going on anymore, I hate using `apt`.

LevitatingBusinessMan, 2020-07-03:
I again confirmed that building via the PKGBUILD I made in the aur I can somehow get PointTracker as the default tracker.
Still the comboboxes don't work on my machine, I also tried switching qt theme. I will now try using a DE

LevitatingBusinessMan, 2020-07-09:
I just compiled all of this on my laptop running Pop! OS, and then manually moved the binaries to my arch installation. And I am getting the same issues as before, so I am totally clueless as to what's happening. And I will be seeing if anyone else with an up-to-date arch installation is facing these issues.

flowtec, 2020-08-01:
@LevitatingBusinessMan 

You got it compiled on Pop!_OS?
With "SDK_WINE=ON"?

I get the same error like you on Pop!_OS 20.4:

[ 93%] Generating opentrack-wrapper-wine.exe.so
/usr/bin/ld: relocatable linking with relocations from format elf64-x86-64 (/usr/lib/x86_64-linux-gnu/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.xGuQNV.o) is not supported
winebuild: /usr/bin/ld failed with status 1
winegcc: /usr/lib/wine/winebuild failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:63: proto-wine/opentrack-wrapper-wine.exe.so] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:2322: proto-wine/CMakeFiles/wine-wrapper.dir/all] Fehler 2
make: *** [Makefile:133: all] Fehler 2

I am trying the hole week on my PC. Then i set up a VM with a fresh setup of Pop!_OS. 
-Tried the wine-setup of the Distribution - no success, like my Main-PC.
wine --version
wine-5.0 (Ubuntu 5.0-3ubuntu1)
-Reverted to a snapshot before setup Wine. Tried Wine with the Source from WineHQ (winehq-devel) - no success.
wine-5.13

This is what i tried and finally i got to 93% - but i can't figure out what to do now. Maybe something is missing?

sudo apt update
sudo apt upgrade
sudo dpkg --add-architecture i386
sudo apt update
sudo apt install wine64 wine32
sudo apt install cmake-curses-gui
sudo apt install qttools5-dev-tools qttools5-dev
sudo apt install libprocps-dev
sudo apt install qtbase5-private-dev
sudo apt install wine64-tools 

git clone https://github.com/opentrack/opentrack
cd opentrack
mkdir build
cd build
ccmake ..

Press 'c'
Set SDK_WINE = ON (per Space)
Press 'c'
Press 'g'

cmake ..
make

BTW:
It compiled successfully, when i set "SDK_WINE=OFF" - but i need it ON to use it with Steam and PointTracker.
I am not an advanced Linux user and just try to avoid to use my old Windows 10 SSD for gaming anymore.
Also had a look at all the other closed and open issues here (599,1082,3129,etc.).
Had also messed up with many trials to install wine32-tools and the dependencies , with no success.

LevitatingBusinessMan, 2020-08-02:
@flowtec, I encountered that same issue as you, but I never fixed it.

I only wanted to test if the menus worked on Pop!OS, so I resorted to building without wine.

sthalik, 2020-08-02:
@flowtec 
> Press 'c'
> Set SDK_WINE = ON (per Space)
> Press 'c'
> Press 'g'

> cmake ..
> make

Did you run `make install`?

LevitatingBusinessMan, 2020-08-02:
@sthtalk, it's a build error...

sthalik, 2020-08-02:
Alright, show me `ldd` on some library that fails to load.

sthalik, 2020-08-02:
> /usr/bin/ld: relocatable linking with relocations from format elf64-x86-64 (/usr/lib/x86_64-linux-gnu/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.xGuQNV.o) is not supported

I see you're using Ubuntu. I've had bad experience with Ubuntu's way of packaging Wine tools, i.e. `winegcc`. Build Wine on your own, or build only the 32-bit version.

> sudo apt install wine64-tools

Or install a version of wine-tools that works with multilib wine.

flowtec, 2020-08-03:
Thank you both for the quick replies.

> ...Worst-case scenario just install wine32-tools.

I tried that - Had also messed up with many trials to install wine32-tools and the dependencies , with no success.
Next i will try to build Wine on my own. Already had a quick look at the WineHQ Wiki about that. Maybe next weekend...

Another thought: If Ubuntu is such bad in packaging Wine tools and i have VMware running - do you could recommend some Distribution for compiling opentrack with Wine-support?

LevitatingBusinessMan, 2020-10-07:
[image]

Issue is still here, after a complete system update.
Still don't know why this is happening.

sthalik, 2020-10-08:
See terminal output.

LevitatingBusinessMan, 2020-10-08:
It's still the same, nothing has changed.
Others on ArchLinux don't have any issues with opentrack, which is also very weird.
I am not sure what makes my machine special here.

If you want I could again put in some log lines?

sthalik, 2020-10-09:
Then how about `LD_DEBUG`?

XenoPL, 2021-03-04:
HI,
I've stumbled upon similar issue. I'm trying to build OpenTrack 2.3.12 for Fedora 33, build proces went fine, but opentrack sees no input trackers nor output protocols. Titlebar says -128-NOTFOUND :: default.ini

[image]

Steps taken with output logs included:
- unpack source tar.gz
 - cd ./opentrack-opentrack-2.3.12
 - mkdir ./build && cd ./build
 - cmake ..
   <cmake output here: https://pastebin.pl/view/fb187e8e >
 - ccmake .
 <
  enabled sdk_wine
  changed intall path to /home/<my_user>/opentrack
  configure
  generate
 >
 - make
  <build log here: https://pastebin.pl/view/4c69f320 >
 - make install
  <intall log here: https://pastebin.pl/view/32e70c6f >
 - cd ~/opentrack
 - ./bin/opentrack

Here's ld output: https://pastebin.pl/view/8c551262

May I ask for 'il of assistance here?

sthalik, 2021-03-04:
Pastebin links broken. Drag-and-drop directly in this text box.

XenoPL, 2021-03-04:
cmake log:
`cmake ..
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- User settings file 'sdk-paths-Xeno@GNU-Linux.cmake' doesn't exist
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3")
-- module api
-- module compat
-- module csv
-- module filter-accela
-- module filter-ewma
-- module filter-hamilton
-- module user-interface
-- module logic
-- module migration
-- module executable
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-udp
-- module qxt-mini
-- module spline
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-hatire
-- module tracker-linux-joystick
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- module video
-- -- Compile flags:
--  CXX: -ggdb -Wall -Wextra -Wpedantic
--  CXX_RELEASE: -O3 -march=native
-- -- Link flags:
--  LINK_SHARED:
--  LINK_SHARED_RELEASE:
-- -- Static archive flags:
--  STATIC:
--  STATIC_RELEASE:
-- --
-- Configuring done
-- Generating done
-- Build files have been written to: /home/Xeno/tmp/opentrack/opentrack-opentrack-2.3.12/build`

Build log:
`Scanning dependencies of target i18n-module-video
[  0%] Generating moc_video-widget.cpp
[  0%] Generating moc_camera.cpp
[  0%] Generating moc_export.cpp
[  0%] Running lupdate for video
[  0%] Built target i18n-module-video
Scanning dependencies of target i18n-module-api
[  0%] Generating moc_plugin-support.cpp
[  0%] Generating moc_export.cpp
[  0%] Generating moc_plugin-api.cpp
[  0%] Running lupdate for api
[  0%] Built target i18n-module-api
Scanning dependencies of target i18n-module-compat
[  1%] Generating moc_variance.cpp
[  2%] Generating moc_activation-context.cpp
[... 1174 lines cut ...]
   1126845:
   1126845:     calling init: /usr/lib64/libbfd-2.35-18.fc33.so
   1126845:
   1126845:
   1126845:     calling init: /usr/lib64/libctf.so.0
   1126845:
   1126845:
   1126845:     initialize program: ld
   1126845:
   1126845:
   1126845:     transferring control: ld
   1126845:
ld: warning: cannot find entry symbol _start; not setting start address
   1126845:
   1126845:     calling fini: ld [0]
   1126845:
   1126845:
   1126845:     calling fini: /usr/lib64/libctf.so.0 [0]
   1126845:
   1126845:
   1126845:     calling fini: /usr/lib64/libbfd-2.35-18.fc33.so [0]
   1126845:
   1126845:
   1126845:     calling fini: /usr/lib64/libdl.so.2 [0]
   1126845:`

sthalik, 2021-03-04:
ARGH! What I meant was drag-attach the file, not the contents.

XenoPL, 2021-03-04:
As requested above, comment edited to somehow more readabale form.

sthalik, 2021-03-04:
You're running `ld` and not `ldd` as you should. You should run `ldd` on a few modules (in `libexec/opentrack`) to see why they're not loading.

XenoPL, 2021-03-04:
Should opentrack main executable be dependent on tracker and proto libs? I can't find any reference to those libs, nor reference to those libs in libs used by main executeable file.
ldd logs:
[libexec_ldd.log](https://github.com/opentrack/opentrack/files/6085060/libexec_ldd.log)
[opentrack_ldd.log](https://github.com/opentrack/opentrack/files/6085061/opentrack_ldd.log)

sthalik, 2021-03-04:
> Titlebar says -128-NOTFOUND :: default.ini

You need git in PATH at build time.

> Should opentrack main executable be dependent on tracker and proto libs?

No.

I also need stderr. `ldd` looks good to me.

sthalik, 2021-03-04:
> unpack source tar.gz

And don't do that, build master directly. Don't use tags or any other branch.

XenoPL, 2021-03-04:
launching opentrack gives me just this one warning to the console:
WARN QXcbConnection: XCB error: 5 (BadAtom), sequence: 552, resource id: 0, major code: 20 (GetProperty), minor code: 0

git should be in PATH at build time, as it resides in /usr/bin
which git
/usr/bin/git
 
 I wanted to build from release tarball, as my next planned step is to make proper fedora package out of it.
 Assuming release contains buildable/working code it should be fine.

XenoPL, 2021-03-04:
Ok, i've cloned source directly from github, build went fine, titlebar now says opentrack-2.3.12-10-g058942f4 :: default.ini
But still no trackers nor output protocols available.
Is there a flag to build it with some more verbose debug messages,  ori  just have to add the debug line from past comments?

sthalik, 2021-03-05:
At this point, run opentrack under strace and see stat64(2)/openat(2) calls after opendir(2) of the library path.

XenoPL, 2021-03-05:
I've run opentrack with strace and it seems there are two ways libs are open:
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-user-interface.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-migration.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-logic.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-spline.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-pose-widget.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-api.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-options.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-compat.so", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/home/Xeno/opentrack/libexec/opentrack/opentrack-qxt-mini.so", O_RDONLY|O_CLOEXEC) = 3

and
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-accela.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-ewma.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-hamilton.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-freepie-udp.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-fusion.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-hatire.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-linux-joystick.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-s2bot.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-test.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-udp.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-fgfs.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-udp.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-wine.so", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9
openat(AT_FDCWD, "/home/Xeno/opentrack/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9

However accela filter seems to be loaded even if it uses same funky path as trackers and protocol libs. Suspicious  part of the log and full strace attached:
[libopen_strace.log](https://github.com/opentrack/opentrack/files/6091409/libopen_strace.log)
[strace.log](https://github.com/opentrack/opentrack/files/6091410/strace.log)

sthalik, 2021-03-05:
So here:

https://github.com/opentrack/opentrack/blob/master/api/plugin-support.hpp#L218

Change to:

```
                if (std::any_of(ret.cbegin(),
                                ret.cend(),
                                [&lib](const std::shared_ptr<dylib>& a) {
                                    return a->type == lib->type && a->name == lib->name;
                                }))
                {
                    if (!(load_mode & dylib_load_quiet))
                        qDebug() << "duplicate lib" << filename << "ident" << lib->name;
                    qDebug() << "loaded" << filename << lib->name << lib->type;
                    continue;
                }
```

The line before `continue`.

sthalik, 2021-03-05:
> However accela filter seems to be loaded even if it uses same funky path as trackers and protocol libs. 

The path is fine, try listing it.

XenoPL, 2021-03-05:
yep ~/opentrack as configured
also app working dir, bin and libexex dirs are in the path

sthalik, 2021-03-05:
Alright, then how about adding that debug statement?

XenoPL, 2021-03-07:
With debug line added in line 254 compilation ends with error: 
 error: no match for ‘operator<<’ (operand types are ‘QDebug’ and ‘dylib_type’)  followed by:
note:   no known conversion for argument 1 from ‘dylib_type’ to ‘QChar’

@[ 64%] Building CXX object logic/CMakeFiles/opentrack-logic.dir/extensions.cpp.o
Full [build.log](https://github.com/opentrack/opentrack/files/6097935/build.log)

sthalik, 2021-03-07:
Then use `(unsigned)lib->type`.

XenoPL, 2021-03-08:
Ok, that worked. So it seems assert in line 246:
if (std::any_of(ret.cbegin(),
                                ret.cend(), 

fails.
I've added debug messages at the top of the for loop (line 238), in the invalid type branch (line 243) and in line 254 as suggested above. All I've got were messages inserted at top of the loop. 
Messages I got on console:

`
[Xeno@localhost opentrack]$ ./bin/opentrack 2>&1
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-accela.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-ewma.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-filter-hamilton.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-freepie-udp.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-fusion.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-hatire.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-linux-joystick.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-s2bot.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-test.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-tracker-udp.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-fgfs.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-udp.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/opentrack/bin/../libexec/opentrack//opentrack-proto-wine.so"
`

XenoPL, 2021-03-08:
To be clear, here's how  lib load loop looks like here after the changes:
`
            for (const QString& filename : dir.entryList({ filter.glob }, QDir::Files, QDir::Name))
            {
                qDebug() << "Loading module: " << QString("%1/%2").arg(library_path, filename);
                dylib_load_mode load_mode_{filter.load_mode | load_mode};
                auto lib = std::make_shared<dylib>(QString("%1/%2").arg(library_path, filename), filter.t, load_mode_);

                if (lib->type == type::Invalid)
                {
                    qDebug() << "Module: invalid type " << filename << lib->name << (unsigned)lib->type;
                    continue;
                }

                if (std::any_of(ret.cbegin(),
                                ret.cend(),
                                [&lib](const std::shared_ptr<dylib>& a) {
                                    return a->type == lib->type && a->name == lib->name;
                                }))
                {
                    if (!(load_mode & dylib_load_quiet))
                        qDebug() << "duplicate lib" << filename << "ident" << lib->name;
                    qDebug() << "loaded" << filename << lib->name << (unsigned)lib->type;
                    continue;
                }

                ret.push_back(std::move(lib));
            }
`

XenoPL, 2021-03-09:
I've made small progress, last build made PointTracker 1.1 input appear, it was opencv devel package missing.
But still no other input, no output and no filters other than Accela are qvailable. SO in short i'm in a place where OP started
I guess I'm missing some more libs?

So far I have intalled:
 - git-2.29.2
 - cmake-3.18.4
 - qt5-qttools-5.15.2
 - qt5-qttools-devel-5.15.2
 - qt5-qtbase-5.15.2
 - qt5-qtbase-devel-5.15.2
 - qt5-qtbase-private-devel-5.15.2-2
 - procps-ng-3.3.16
 - procps-ng-devel-3.3.16
 - opencv-4.3.0
 - opencv-devel-4.3.0
 - libevdev-1.11
 - libevdev-devel-1.11
 - wine-6.3
 - wine-devel-6.3

LevitatingBusinessMan, 2021-03-09:
For reference, these are all packages I have installed on my system: https://catbin.xyz/BICUg.

sthalik, 2021-03-10:
I've made a mistake, the debug statement saying "loaded" needs to go below `continue;` and the closing curly brace.

XenoPL, 2021-03-10:
Ok, here's the console output:
`
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-filter-accela.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-filter-accela.so" "Accela" 3735927486
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-filter-ewma.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-filter-ewma.so" "EWMA" 3735927486
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-filter-hamilton.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-filter-hamilton.so" "Hamilton" 3735927486
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-filter-kalman.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-filter-kalman.so" "Kalman" 3735927486
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-aruco.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-aruco.so" "aruco -- paper marker tracker" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-easy.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-easy.so" "Easy Tracker 1.1" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-freepie-udp.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-freepie-udp.so" "FreePIE UDP receiver" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-fusion.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-fusion.so" "Fusion" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-hatire.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-hatire.so" "Hatire Arduino" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-hydra.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-hydra.so" "Razer Hydra -- inertial device" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-linux-joystick.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-linux-joystick.so" "Linux Joystick input" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-pt.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-pt.so" "PointTracker 1.1" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-s2bot.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-s2bot.so" "S2Bot receiver" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-steamvr.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-steamvr.so" "Valve SteamVR" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-test.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-tracker-test.so" "Test tracker" 3405692655
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-udp.so"
[... cut ...]
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-proto-wine.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-proto-wine.so" "Wine -- Windows layer for Unix" 3735941133
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:240]: Loading module:  "/home/Xeno/tmp/opentrack/github/opentrack/build/install/bin/../libexec/opentrack//opentrack-video-opencv.so"
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/api/plugin-support.hpp:262]: loaded "opentrack-video-opencv.so" "opencv" 195948557
`

XenoPL, 2021-03-10:
BTW could it be some difference between Qt 5.15 and 5.14? I've tried opentrack in  SUSE Tumbleweed, probably build with Qt 5.14 and it works, but opentrack installed from rpm made from SUSE spec under Fedora with Qt 5.15  is broken.
Or maybe Fedora is doing something stupid ?

XenoPL, 2021-03-15:
I've compared suse and fedora ldd output for tracker and proto libs, mst obvious thing is fedora libs lack dependance on 
 - /usr/lib64/libdouble-conversion.so.3 (double-conversion, double-conversion-devel)
 - /usr/lib64/libcap.so.2 (libcap, libcap-devel)
 - /usr/lib64/libselinux.so (libselinux, libselinux-devel)
Although I do have installed proper packages that provide said libs and their headers.

What else can I do to help diagnose this problem? Any debug messages can be added to the libs themselves to see why they are failing to load properly?

sthalik, 2021-03-15:
Add some more debug info on <<https://github.com/opentrack/opentrack/blob/master/opentrack/main-window.cpp#L86>>. Before `connect(` on an empty line:

```c++
qDebug() << "tracker list" << modules.trackers.size() << ui.iconcomboTrackerSource->count();
```

XenoPL, 2021-03-15:
Pasted in line 86:
`
    for (dylib_ptr& x : modules.filters())
        ui.iconcomboFilter->addItem(x->icon, x->name, x->module_name);

    qDebug() << "tracker list" << modules.trackers.size() << ui.iconcomboTrackerSource->count();
    connect(ui.iconcomboTrackerSource, &QComboBox::currentTextChanged,
`

Compilation fails with error:
`
opentrack/main-window.cpp:86:43: error: invalid use of member function ‘Modules::dylib_list& Modules::trackers()’ (did you forget the ‘()’ ?)
   86 |     qDebug() << "tracker list" << modules.trackers.size() << ui.iconcomboTrackerSource->count();
`

XenoPL, 2021-03-15:
Ok, changed offending line as sugessted by err description
`
    qDebug() << "tracker list" << modules.trckers().size() << ui.iconcomboTrackerSource->count();
`

Opentrack compiled, new line in debug output is:
DEBUG [/home/Xeno/tmp/opentrack/github/opentrack/opentrack/main-window.cpp:86]: tracker list 9 9

[14 more replies]