opentrack/opentrack #1117: How to install on ubuntu

nkv123, 2020-07-16:
Hello everyone, i am using ubuntu 18.04
Are there any installation instructions for ubuntu?
Is there a ppa, docker or snap for this?

Attigliuzzo, 2020-07-29:
Feel free to test my debian package made for Debian testing. You will find it in this place.

wleoncio, 2020-07-31:
There's a snap package available [here](https://snapcraft.io/opentrack), but it's quite outdated (from 2016).

p-s-t, 2020-12-03:
Hello, thank you all for bringing open track to linux.

I would like to request a hint if somebody can help please.
I apologize for my dumb questions in advance.

Not a programmer, nor do i have any knowledge on how to do all these Build making and compiling.
But i tried to follow the opentrack wiki "Building on Linux"

i did "compiled" i guess and installed openCV 3.4.12 which i gathered supposed to be required.
It does seem like it was finished successfully according to guide.
https://linuxize.com/post/how-to-install-opencv-on-ubuntu-18-04/

Then i preceded with all the steps from wiki
and it also seemed like a success.

I now currently have in my home directory(or folder) 
opentrack/install
with most likely the install  "build" i guess :-)

But then the wiki ends
with :
"The resulting build output will be placed in the install/ directory. It will not 'install' itself anywhere outside of the current directory."

So i would like to ask how do you install it. This must sound funny to you but i have no clue. :-)
I was considering just using 
sudo apt install opentrack
but i think that would just install some who knows what version from the internet and not from the install folder i just obtained, right ?

also 
instruction from suggested site FliersHaveMerit seem to be contradictory 
specifically  /dev/uinput directory creating part - that is strange because i already have file named uinput in dev.
Which results in system not letting me create folder with same name. 
And it is fairly old guide anyway i have a feeling it might not be what i need.

Attigliuzzo, 2020-12-03:
@p-s-t 
if you build successfully binaries, enter in the install folder i suppose `/your path/install/opentrack/bin or /usr/local/opentrack/lib` and from terminal type `./opentrack`.
I built a deb package for my debian, if you want to install it you can use gdebi with its graphical frontend (right click-open with) or type in the folder where it is `dpkg -i opentrack.xx.yy.zz.deb` as root.

p-s-t, 2020-12-05:
Thank you for your answer
It was this part:

"/your path/install/opentrack/bin from terminal type ./opentrack"

After that open track just launched :) I feel silly i did not just try that. Did not occurred to me it was executable.
It  is just missing freetrack output, but that is another issue which i will try to figure out around here or ask in appropriate topic.
Thank you very much good sir.

But please don't run away yet. Can we solve this for others too.
How did this happen: 
That line on wiki is very misleading. Second sentence especially.
"The resulting build output will be placed in the install/ directory. It will not 'install' itself anywhere outside of the current directory."
I assumed that there are more steps to actually finish it.

Can i please request change of this wiki line to something like:

"The resulting build output will be placed in the install/ directory. 
You can now run it by launching opentrack file. 

from terminal type ./opentrack"

Attigliuzzo, 2020-12-05:
@p-s-t 
The freetrack plugin is only for windows, under own fabulous OS you can choose from: easytracker or pointTracker (IR leds), hatire (arduino), Freepie (smartphone), UDP (over web).
Feel free to edit/improve the wiki by yourself.

sthalik, 2020-12-05:
> The freetrack plugin is only for windows, under our fabulous OS you can choose from: easytracker or pointTracker (IR leds), hatire (arduino), Freepie (smartphone), UDP (over web).

Freetrack is an output plugin. Its functionality on Linux is entirely replaced by the Wine plugin (also supports the X-Plane plugin).

p-s-t, 2020-12-06:
I think i will just continue here
Thank you for all the answers 
I was not able to figure out a way to enable the other outputs only ones present are 
Flight gear , libdev joystick reciever , UDP over network
no xplane nor wine outputs are present

Got a feeling this needs to be somehow specified when compiling. Is that correct ?

Also i did tried that .deb package you mentioned. In hope to find it included in that one.
Installs nicely, When launched from start menu after a short delay it just closes without any error.
When i found where it installed itself 
/usr/local/opentrack
and launched it from terminal it reveals it needs Qt 5.15 to work.
So i set out to install that, which from this very version requires special online installer and registration.
After that and by using command to point it towards newly instlled Qt version
LD_LIBRARY_PATH=/opt/Qt/5.15.2/gcc_64/lib ./opentrack
it worked however still just those three output methods.

rackhamlerouge9, 2021-07-17:
Dear all, thanks for all your work in writing and testing this software.

Please may I add, as a linux novice, that having spent a half day following various guides trying to compile this (heaps of hurdles; realising I needed cmake not make, installing Qt5, etc, etc, still getting nowhere), I would deeply appreciate a step-by-step idiot's-guide to compiling this source code on linux, then installing it.

My deepest thanks to @Attigliuzzo, whose deb package I shall now try, having found this discourse!

My thanks again.

Attigliuzzo, 2021-07-18:
Thank  @rackhamlerouge9 for testing my package, i have not compiled it with the wine plugin because it requires to install 32bit stuffs. Anyway you can follow my guide:

> 1- install git, make a dir and launch:  git clone https://github.com/opentrack/opentrack
2- install following packages: gcc9, gcc9-base, gcc9-multilib,lib32gcc-8-dev,cmake, cmake-curses-gui, gist, libqt5concurrent5, libqt5core5a, libqt5dbus5, libqt5gui5, libqt5network5, libqt5printsupport5, libqt5sql5, libqt5test5, libqt5widgets5, libqt5xml5, libevdev-dev, libevdev2, libeigen3-dev, libopenCV-dev,qtbase5-private-dev, libqt5serialport-dev, g++9, g++ multilib, qt5base-dev, qttools-dev, libprocps-dev.
3- create a directory in ../opentrack/build, then cd build
4- ccmake ..  will tell about missing deps. install any missing deps, empty the build folder and re-launch ccmake ..
5- press "c" to to continue and then "t", edit any items like the install path:
 SDK_ENABLE_LIBEVDEV            on         ---> not available, already integrated
CMAKE_INSTALL_PREFIX          /usr/local/opentrack
6- cmake .
7- make -j5 
8- make install

It isn't updated but it will be a good starting point, to build your package you can use checkinstall.
Regards

rackhamlerouge9, 2021-07-19:
> Thank ***@***.***(https://github.com/rackhamlerouge9) for testing my package, i have not compiled it with the wine plugin because it requires to install 32bit stuffs. Anyway you can follow my guide:

Many thanks @Attigliuzzo

For reference, I'm running Lubuntu 20.04.2 on a Teclast F6 Pro laptop.

I'm dreadfully sorry I didn't know how to get or use ccmake, so I just use apt-get install on your package, and when it told me a dependency I searched online for it:

Your package needed:
https://software.pureos.net/package/bin/landing/libopencv-video4.5which required me to install
https://software.pureos.net/package/bin/landing/libopencv-dnn4.5
https://software.pureos.net/package/bin/landing/libprotobuf23
https://software.pureos.net/package/bin/landing/libopencv-calib3d4.5
https://software.pureos.net/package/bin/landing/libopencv-features2d4.5
https://software.pureos.net/package/bin/landing/libopencv-flann4.5https://software.pureos.net/package/bin/landing/libopencv-features2d4.5
https://software.pureos.net/package/bin/landing/libopencv-imgproc4.5
https://software.pureos.net/package/bin/landing/libopencv-core4.5

I include the links here because I could not find them in my default repository. It was a bit of a slog, but using the "sudo apt-get install" command it told me at each stage the needed dependancies until I had the above list.

Following your instructions I got:
Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/x11-keymap.cpp.o
/home/al/Downloads/opentrack-opentrack-2.3.13/qxt-mini/qxtglobalshortcut_x11.cpp:40:10: fatal error: qpa/qplatformnativeinterface.h: No such file or directory
40 | #include <qpa/qplatformnativeinterface.h>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I did try installing your .deb file, but while the opentrack icon appeared, it would not run.

I'm very much open to suggestions if you have any, but most of all I'm grateful someone else thinks there needs to be a debian package!

Yours sincerely and with thanks,
Al

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Attigliuzzo, 2021-07-20:
Hi @rackhamlerouge9,
despite ubuntu is a folk of Debian isn't always possible to satisfy cross dependecies, the use of ccmake is simple it is useful to configure your system before compile your code. 
Without right libraries installed my opentrack won't run.

I read you miss the header "qplatformnativeinterface.h", so searching from the main [site of ubuntu packages](https://packages.ubuntu.com/search?mode=exactfilename&suite=hirsute&section=all&arch=any&keywords=qplatformnativeinterface.h&searchon=contents) you will need to install the "qtbase5-private-dev" package.

If you don't want to waste your time, i remember that in the benchmark sim forum a guy built opentrack for ubuntu. This is [the page](https://www.benchmarksims.org/forum/showthread.php?16825-ANN-opentrack-2-0-beta-1-released!&p=558250#post558250), and this is the [direct link](https://www.magentacloud.de/lnk/jwBpmS02) to the file. Another guide [here](https://www.benchmarksims.org/forum/showthread.php?41827-UNOFFICIAL-Running-BMS-on-Linux-WINE-with-opentrack-HOWTO).
Regards

rackhamlerouge9, 2021-07-20:
@Attigliuzzo ,

You are a saint! I shall try again as you advise and failing that will go down the route of the package and guide to which you so kindly linked. Unfortunately I'm on nights this week so this will have to be a job for next weekend. I hope that when it works I can make a Lubuntu bootable USB with Flightgear installed on it and configured to connect to a server, and post it with a joystick and head-tracker to a pilot friend so he can teach me to fly via the internet. There is such potential still lying dormant at the moment!

Thank you again,
Al

Sent from ProtonMail mobile

rackhamlerouge9, 2021-07-20:
@nkv123 ,

I'm so sorry, I am replying by email and had thought @attigluzzio had written your very kind and patient guidance! You were absolutely right, of course, about my needing to learn of the ubuntu packages guide. Mind: blown. Thank you so much :D

Al

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐