opentrack/opentrack #1132: Camera not stopping with LibUSB

blueeagle69, 2020-08-20:
Hi.

I just installed the LibUSB driver instead of the CL Eye driver.
When I stop the tracking, the camera still appears to be active.
(Red light on camera)
(Windows 10 64).
Thanks.

aljazzair, 2020-09-23:
Hello
Thank you for creating and maintaining this project.
I have also tried PS3 eye with opentrack and ever since I installed it , the red led does not come off until the USB plug is physically removed. 
Was going to file a bug until I saw this.
It would be beneficial to have a functioning light indicating whether the camera is on or not.

sthalik, 2020-09-24:
Already done in ac056aa85dca83be3b1a14df7b20fd07104e052c. I'll be releasing a test build.