opentrack/opentrack #1136: Linux: opencv2/core.hpp: No such file or directory

theluckyone17, 2020-08-23:
I've recently given up on LinuxTrack for head tracking purposes (running X-Plane 11), and I'd like to move to OpenTrack. Compiling it manually on my Gentoo system fails:

> Scanning dependencies of target opentrack-video-opencv
> [ 73%] Building CXX object video-opencv/CMakeFiles/opentrack-video-opencv.dir/impl-camera.cpp.o
> In file included from /home/noel/src/opentrack/video-opencv/impl-camera.cpp:1:
> /home/noel/src/opentrack/video-opencv/impl.hpp:14:10: fatal error: opencv2/core.hpp: No such file or directory
>    14 | #include <opencv2/core.hpp>
>       |          ^~~~~~~~~~~~~~~~~~
> compilation terminated.

The header file does appear to be installed on my system, provided by OpenCV version 4.1.2, located at: _/usr/include/opencv4/opencv2/core.hpp_, I imagine for backwards compatibility purposes. 

Can anyone assist in pointing the build process to the correct location for that header file?

sthalik, 2020-08-23:
See `OpenCV_DIR`.

theluckyone17, 2020-08-24:
That didn't do it, unfortunately. CCMake will allow me to overwrite the existing _/usr/local/share/OPenCV_ with _/usr/include/opencv4/_, but it won't generate a configuration file with that entry made. Tapping 'c' to configure overwrites the include directory with the original. There doesn't seem to be variable to point to the OpenCV include files, like there is for Eigen3 (EIGN3_INCLUDE_DIR). 

That said... I was able to find a Gentoo overlay (ace) that provides an ebuild for OpenTrack, but it generates a similar error, with or without the opencv use keyword added: 

> FAILED: cv/CMakeFiles/opentrack-cv.dir/affine.cpp.o
>/usr/bin/x86_64-pc-linux-gnu-g++ -DBUILD_CV -DOPENTRACK_ORG=\"opentrack-2.3\" -DQT_CORE_LIB -DQT_GUI_LIB ->DQT_MESSAGELOGCONTEXT -DQT_NETWORK_LIB -DQT_NO_DEBUG ->DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_WIDGETS_LIB -Icv -I/var/tmp/portage/games-util/opentrack->2.3.11/work/opentrack-opentrack-2.3.11/cv -I/var/tmp/portage/games-util/opentrack-2.3.11/work/opentrack-opentrack-2.3.11 -I. ->I/var/tmp/portage/games-util/opentrack-2.3.11/work/opentrack-opentrack-2.3.11/cv/SYSTEM -isystem /usr/local/include/opencv >-isystem /usr/include/qt5 -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem >/usr/include/qt5/QtGui -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtNetwork  -DNDEBUG -ggdb -Wall ->Wextra -Wpedantic -fPIC -fvisibility=hidden   -fuse-cxa-atexit -fno-plt -fPIC -std=c++17 -MD -MT cv/CMakeFiles/opentrack->cv.dir/affine.cpp.o -MF cv/CMakeFiles/opentrack-cv.dir/affine.cpp.o.d -o cv/CMakeFiles/opentrack-cv.dir/affine.cpp.o -c >/var/tmp/portage/games-util/opentrack-2.3.11/work/opentrack-opentrack-2.3.11/cv/affine.cpp
>In file included from /var/tmp/portage/games-util/opentrack-2.3.11/work/opentrack-opentrack-2.3.11/cv/affine.cpp:8:
>/var/tmp/portage/games-util/opentrack-2.3.11/work/opentrack-opentrack-2.3.11/cv/affine.hpp:10:10: fatal error: >opencv2/core.hpp: No such file or directory

And the odd thing is, it's pulling the OpenCV include directory from _/usr/local/include/opencv_... which may have been left over when I was trying to install LinuxTrack, years ago. Removing that directory (and the rest of the local install), allowed both the ebuild version and the manual build to complete successfully.

Chalk this one up to user error. :D

closed 2020-08-24 by theluckyone17