opentrack/opentrack #1137: Virus signature matching in opentrack-video-opencv.dll?

epmoyer, 2020-08-24:
In release 2.3.12 I am seeing Windows Defender identify `Trojan:Win32/Wacatac.D5!ml` as being present in `C:\Program Files (x86)\opentrack\modules\opentrack-video-opencv.dll`

Perhaps this is a false positive, or perhaps the release is infected?  Can anyone else scan 2.3.12 and see if it comes up clean?  The file has been quarantined on my machine and I see nothing else in a full system scan.

epmoyer, 2020-08-24:
I ran the 2.3.12 release installer (the .exe version) on a sandboxed vm and got the same result, so Windows defender is definitely matching the .dll (falsely or not) for the Wacatac malware.

logans0401, 2020-08-24:
Confirmed, in opentrack-2.3.12-win32-setup.exe, Windows Defender currently detects \opentrack\modules\opentrack-video-opencv.dll as a Wacatac trojan and is automatically quarantined/deleted upon installation. When this occurs, some functionality of opentrack is removed, such as webcam detection, and access to certain properties of your webcam is removed.

Virus-total: https://www.virustotal.com/gui/file/5fc4277fd2afc253d645aeaa9d37b42588bc318eec8adfeabb57b56960773fed/detection

AbyssalReClass, 2020-08-24:
I checked that file in the portable version, Windows Defender flags it as well. Same virustotal result as above. It's an issue with the DLL.

marvk, 2020-08-24:
I'm having the same issue, the 2.3.12 installer is also recognized to be malicious by Windows Defender. [Virus Total score is 2/67,](https://www.virustotal.com/gui/file/0b92b3ae911d88c7be056f296fe937df73a9b36d306f8c498948dd29c5ffd447/detection) so I feel like it's unlikely to be an issue, but I'm still not using it right now.

sthalik, 2020-08-24:
> Perhaps this is a false positive, or perhaps the release is infected?

The release is clean. Thanks for the report.

truemetro, 2020-08-24:
same issue here. I've just quarentined it for now, hopefully it's a false positive but I'm not interested in taking that chance haha.

sthalik, 2020-08-25:
I reported this via Microsoft's form. 

> We have removed the detection.  Please follow the steps below to clear cached detection and obtain the latest malware definitions.
>     1. Open command prompt as administrator and change directory to c:\Program Files\Windows Defender 
>     2. Run “MpCmdRun.exe -removedefinitions -dynamicsignatures”
>     3. Run "MpCmdRun.exe -SignatureUpdate"

Or alternately, wait a while for the definitions to update on their own.

marvk, 2020-08-25:
@sthalik I can confirm that after taking the steps you described and redownloading the installer, Windows Defender is no longer reporting any threats!