opentrack/opentrack #1138: the Opentrack 2.3.12 crashes when the window is closed

ljgdasfhk, 2020-08-25:
when i click close window button the program crashes
`Сигнатура проблемы:
  Имя события проблемы:	BEX
  Имя приложения:	opentrack.exe
  Версия приложения:	0.0.0.0
  Отметка времени приложения:	5f3e37fd
  Имя модуля с ошибкой:	opentrack-tracker-wii.dll_unloaded
  Версия модуля с ошибкой:	0.0.0.0
  Отметка времени модуля с ошибкой:	5f3e3852
  Смещение исключения:	6d9b9b32
  Код исключения:	c0000005
  Данные исключения:	00000008
  Версия ОС:	6.1.7601.2.1.0.256.1
  Код языка:	1049
  Дополнительные сведения 1:	0a9e
  Дополнительные сведения 2:	0a9e372d3b4ad19135b953a78882e789
  Дополнительные сведения 3:	0a9e
  Дополнительные сведения 4:	0a9e372d3b4ad19135b953a78882e789

Ознакомьтесь с заявлением о конфиденциальности в Интернете:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0419

Если заявление о конфиденциальности в Интернете недоступно, ознакомьтесь с его локальным вариантом:
  C:\Windows\system32\ru-RU\erofflps.txt
`
Opentrack v.2.3.11 --- Ok.

sthalik, 2020-08-25:
Send me a minidump. Instructions: https://www.wintellect.com/how-to-capture-a-minidump-let-me-count-the-ways/

ljgdasfhk, 2020-09-07:
[opentrack-minidump.zip](https://github.com/opentrack/opentrack/files/5184092/opentrack-minidump.zip)

sthalik, 2024-06-07:
I don't think this is the issue anymore (please respond otherwise).

closed 2024-06-07 by sthalik

ljgdasfhk, 2024-06-07:
This problem has been gone for a long time now. I don't think I should have closed it myself when it stopped appearing. Correct me if I'm wrong about this.