opentrack/opentrack #1139: WiiPointTracker?

lizaoreo, 2020-08-25:
I apologize for asking what I think is probably a dumb question, but how do I get the WiiPointTracker to show up as an input option?  I have tried researching, but people are mostly talking about how to get the LEDs right or how to pair the Wiimote, etc.  I've not found anyone explain how to get the input to appear in OpenTrack.  I tried opening up the tracker-wii folder in the source folder with Visual Studio and building that, but that didn't work either.

lizaoreo, 2020-08-26:
I downgraded from opentrack 2.3.12 to 2.3.11 and now it shows up, not sure if it was removed for some reason or if it's a bug, but I am good.

sthalik, 2020-08-26:
Which Windows version is it?

lizaoreo, 2020-08-26:
No, Windows 10, 2004.

sthalik, 2020-08-27:
Show me the output of Dependency Walker (32-bit version) ran on `modules/opentrack-tracker-wii.dll`. A screenshot of the left window pane will suffice.

lizaoreo, 2020-08-28:
Sure, not sure exactly what you needed, never used this before.

[image]

sthalik, 2020-09-05:
Click on the top level `[-]` twice to collapse the list in the top left corner.

lizaoreo, 2020-09-10:
[image]

sthalik, 2020-09-15:
Alright. No problems there. Open opentrack through Windows console (just drag the executable onto the console window). At startup it's going to load plugins and complain if any of them fail. What's the reason for tracker-wii not loading?

AchilleGR, 2020-09-17:
Sorry for barging in, I have the same issue with version 2.3.12. Version 2.3.11 works fine. Submitting the info you requested @sthalik
```
C:\Program Files (x86)\opentrack>opentrack.exe

C:\Program Files (x86)\opentrack>
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:253]: duplicate lib "opentrack-tracker-rift.dll" ident "Oculus Rift runtime 1.4.0 -- HMD"
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "wii" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-wii.dll: A dynamic link library (DLL) initialization routine failed."
WARN DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="MS Shell Dlg 2", pointsize=7.875, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("DejaVu Sans", lfWidth=0, lfHeight=-11) dpi=192
```

Managed to hunt down the report from event logger also.
```
Version=1
EventType=APPCRASH
EventTime=132448306924516279
ReportType=2
Consent=1
UploadTime=132448306929823632
ReportStatus=268435456
ReportIdentifier=a12d8763-4252-4c5b-bb27-2b40e54062f8
IntegratorReportIdentifier=a72a9359-a97a-46a5-9a03-14cf26dc5e88
Wow64Host=34404
Wow64Guest=332
NsAppName=opentrack.exe
AppSessionGuid=00003334-0001-0010-0695-798c088dd601
TargetAppId=W:0006ff002043ddad3226df9e102729a8ba2c0000ffff!000097caf45713aebe50de1720d3ea0c9fff3d0eb4c8!opentrack.exe
TargetAppVer=2020//08//20:08:44:45!70810!opentrack.exe
BootId=4294967295
ServiceSplit=1977614336
TargetAsId=1343
IsFatal=1
EtwNonCollectReason=1
Response.BucketId=b1d533d5d51709ea8fcfde73a056a1cf
Response.BucketTable=1
Response.LegacyBucketId=2292295323547509199
Response.type=4
Sig[0].Name=Application Name
Sig[0].Value=opentrack.exe
Sig[1].Name=Application Version
Sig[1].Value=0.0.0.0
Sig[2].Name=Application Timestamp
Sig[2].Value=5f3e37fd
Sig[3].Name=Fault Module Name
Sig[3].Value=opentrack-tracker-wii.dll
Sig[4].Name=Fault Module Version
Sig[4].Value=0.0.0.0
Sig[5].Name=Fault Module Timestamp
Sig[5].Value=5f3e3852
Sig[6].Name=Exception Code
Sig[6].Value=c0000005
Sig[7].Name=Exception Offset
Sig[7].Value=00001025
DynamicSig[1].Name=OS Version
DynamicSig[1].Value=10.0.18363.2.0.0.256.48
DynamicSig[2].Name=Locale ID
DynamicSig[2].Value=1032
DynamicSig[22].Name=Additional Information 1
DynamicSig[22].Value=ef36
DynamicSig[23].Name=Additional Information 2
DynamicSig[23].Value=ef367b5c357304155eeff192779a590a
DynamicSig[24].Name=Additional Information 3
DynamicSig[24].Value=29fb
DynamicSig[25].Name=Additional Information 4
DynamicSig[25].Value=29fb646949a1d1ddf8712a39c7b85ff0
UI[2]=C:\Program Files (x86)\opentrack\opentrack.exe
LoadedModule[0]=C:\Program Files (x86)\opentrack\opentrack.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\WINDOWS\System32\KERNEL32.DLL
LoadedModule[3]=C:\WINDOWS\System32\KERNELBASE.dll
[... 182 lines cut ...]
OsInfo[27].Key=wpbuildmaj
OsInfo[27].Value=0
OsInfo[28].Key=wpbuildmin
OsInfo[28].Value=0
OsInfo[29].Key=osver
OsInfo[29].Value=10.0.18362.1082.amd64fre.19h1_release.190318-1202
OsInfo[30].Key=buildflightid
OsInfo[30].Value=58798900-3ce6-45a7-ae1e-a2df8ca981d1
OsInfo[31].Key=edition
OsInfo[31].Value=Professional
OsInfo[32].Key=ring
OsInfo[33].Key=expid
OsInfo[34].Key=containerid
OsInfo[35].Key=containertype
OsInfo[36].Key=edu
OsInfo[36].Value=0
FriendlyEventName=Stopped working
ConsentKey=APPCRASH
AppName=opentrack.exe
AppPath=C:\Program Files (x86)\opentrack\opentrack.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=06637669A136D8B768DD4BBFAC02DF4E
MetadataHash=1896899945
```

lizaoreo, 2020-09-18:
Thanks @AchilleGR, sorry, I've been distracted :)  I got the smooth track app or whatever it is on my iPad now which uses the UDP input for Opentrack, but I'll assist in any troubleshooting.  I assume what Achille provided is all you need for the moment though.

sthalik, 2020-09-21:
I'm going to fix it once I install Windows 10. Server editions don't have `bthprops.cpl` and I can't even load the module.

cpuwolf, 2020-10-23:
I have tested both versions on Win10
 
- 2.3.12 wiiPointerTracker no load
- 2.3.11 wiiPointerTracker works good

sthalik, 2020-10-28:
Can you send me 32-bit bthprops.cpl so I can test it on Windows? Thanks.

cpuwolf, 2020-10-31:
@sthalik [

](url)

cpuwolf, 2020-10-31:
[Uploading bthprops.zip…]()

sthalik, 2020-10-31:
This didn't upload fully. Can you give me a backtrace when loading the module? The .pdb files are in the `debug/` subdirectory.

gconry18, 2020-12-18:
This error is still happening and causing a long delay to starting any tracking. Removing `opentrack-tracker-wii.dll` doesn't seem to fix the issue. It's strange because it was all working perfectly fine a few days ago and nothing in my setup has changed.

Tried with 2.3.12 installed and portable

ultraheroe, 2021-01-17:
Same problem here using the 2.3.12 installed version. (2.3.11 loads the wiimote module well)

[image]

if it helps, I show you the .net framework installed in my W10

[image]

I don't know how, but if I may help you debugging the problem I can send .pdb files or whatever is needed.

Thanks for opentrack!