opentrack/opentrack #1140: Tracker Remains Offline

theluckyone17, 2020-08-27:
After resolving my earlier issue (pointing the build to the right OpenCV version), I have successfully built OpenTrack. In using the application, I've been able to configure the PointTracker 1.1 input, previewing the webcam video properly and tweaking the camera settings so my TrackHat cap shows up properly.

However, after hitting the Start button on the main window, no preview shows up in the main window (even after stopping the preview from the settings/"V4L2 Test Bench" window). In the PointTracker Settings window, the Camera Info: status line remains "Tracker offline" throughout my testing. The camera will not show an active light.

After posting on the X-Plane.org forums, a suggestion was made to build in SDK_WINE support; I did so, with no change in symptoms, but did note that the "X-Plane" output disappeared and was replaced with the Wine output. I thought that weird, but regardless, it did not change the original symptoms.

I've attached my build outputs, both with SDK_WINE=ON and SDK_WINE=OFF, on the off chance they're helpful:
[SDKWINEOFF.ccmake.txt](https://github.com/opentrack/opentrack/files/5133325/SDKWINEOFF.ccmake.txt)
[SDKWINEOFF.cmake.output.txt](https://github.com/opentrack/opentrack/files/5133327/SDKWINEOFF.cmake.output.txt)
[SDKWINEOFF.make.output.txt](https://github.com/opentrack/opentrack/files/5133329/SDKWINEOFF.make.output.txt)
[SDKWINEOFF.make.install.output.txt](https://github.com/opentrack/opentrack/files/5133330/SDKWINEOFF.make.install.output.txt)
[image]

[SDKWINEON.ccmake.txt](https://github.com/opentrack/opentrack/files/5133331/SDKWINEON.ccmake.txt)
[SDKWINEON.cmake.output.txt](https://github.com/opentrack/opentrack/files/5133332/SDKWINEON.cmake.output.txt)
[SDKWINEON.make.output.txt](https://github.com/opentrack/opentrack/files/5133333/SDKWINEON.make.output.txt)
[SDKWINEON.make.install.output.txt](https://github.com/opentrack/opentrack/files/5133334/SDKWINEON.make.install.output.txt)
[image]

theluckyone17, 2020-08-29:
Been doing some poking:
Threw some simple printf statements in _ftnoir_tracker_pt.cpp_ to see if the functions were being accessed. They are. In the order they get called:
- set_fov
- start_tracker
- Tracker_PT
- maybe_reopen_camera
- data ... which continues to be called 'til I hit the stop button.

I don't know enough about debugging (and using GDB) to step through each function and look for anything out of the ordinary... but I do know enough to check and see if the camera's device file is being accessed. I know if I access the tracker settings, then the Camera settings, I can hit the "Start Capture" button, and it'll access the webcam properly. So it works as a control... and sure enough, when I do, I've got the **qv4l2** process locking the device:
```
**Output of lsof | grep video0 :**
qv4l2      3644                       noel  mem       CHR               81,0                4535 /dev/video0
qv4l2      3644                       noel   11u      CHR               81,0       0t0      4535 /dev/video0
qv4l2      3644  3645 QXcbEvent       noel  mem       CHR               81,0                4535 /dev/video0
qv4l2      3644  3645 QXcbEvent       noel   11u      CHR               81,0       0t0      4535 /dev/video0
qv4l2      3644  3646 QDBusConn       noel  mem       CHR               81,0                4535 /dev/video0
qv4l2      3644  3646 QDBusConn       noel   11u      CHR               81,0       0t0      4535 /dev/video0
```
If I then stop the capure, close the settings windows, return to the main window, and hit the Start button... no process locks /dev/video0. So there's got to be a difference between how the camera is being accessed in the Settings window versus the main process.

sthalik, 2020-08-30:
> after hitting the Start button on the main window, no preview shows up in the main window (even after stopping the preview from the settings/"V4L2 Test Bench" window). In the PointTracker Settings [...]

Did you select the right camera in tracker settings? What's the opencv version?

sthalik, 2020-08-30:
```
video-opencv/impl-camera.cpp:36:    cap.emplace(idx + video_capture_backend);
```

Replace with

```
cap.emplace(idx);
```

theluckyone17, 2020-08-30:
OpenCV is version 4.1.2. I've only the one webcam (keeping it simple), and only one shows up in the settings:
[image]

Made the change:
`bool cam::start(info& args)
{
    stop();
    // cap.emplace(idx + video_capture_backend);
    cap.emplace(idx);
`
Rebuilt OpenTrack (_make clean && make -j5 && sudo make install_)

No change in symptoms:
[image]

For giggles, I opened up the PointTracker settings window, the Camera Settings window, and started a preview:
[image]

theluckyone17, 2020-08-30:
OOOOOOhhhh.... update:

Got to thinking about your mention of the OpenCV version... and I had manually built a separate copy of OpenCV version 3.4.11 when I rebuilt LinuxTrack a night or two ago, dumping it in _/usr/local_ to keep it separate from the version installed by the system's package manager.

Reverted the code change in _impl-camera.cpp_ you mentioned above, then rebuilt OpenTrack against OpenCV v3.4.11, and it works fine:
[image]

Looks like it doesn't play nicely with 4.1.2!